chirpstack/chirpstack-concentratord #1: How i change reset pin in config?

erdmanncross, 2020-02-14:
I'm trying to use the Gateway-OS with raspberry-pi and a concentrator from a company in Brazil, the concentrator has the same configuration as RHF0M301, the only difference from the reset pin.
RHF0M301 reset pin 24
Radioenge concentrator = reset pin 26
I would like to know if I can change the settings to use pin 26.
this concentrator has gps:
GPS RXD - UART TX 8
GPS TXD - UART RX 10
and I'm trying to make it work with Gateway-OS.

brocaar, 2020-02-17:
This the the concentratord repo, the concentrator reset is not handled by concentratord but by a Gateway OS start script, see https://github.com/brocaar/chirpstack-gateway-os/blob/master/layers/targets/meta-raspberrypi/recipes-core/gateway-config/files/gateway-config.sh#L273.

closed 2020-02-17 by brocaar

erdmanncross, 2020-02-18:
Sorry, I made some changes to gateway-config.sh adding support for the RadioEnge (BR) gateway.
It is working well with Gateway-OS.
How can I submit changes for your review?

brocaar, 2020-02-19:
@erdmanncross there are probably two things to do:

- [ ] Add the shield configuration to this repository. Instead of using the _generic_ model, it is better to use the calibration values provided by RadioEnge (these are vendor / model dependent). Example: https://github.com/brocaar/chirpstack-concentratord/blob/master/chirpstack-concentratord-sx1301/src/config/vendor/imst/ic880a_eu868.rs

- [ ] Modify the Gateway OS `gateway-config.sh` script to add this gateway to the list

The easiest way to submit modifications for review is to create a pull-request for both modifications.

erdmanncross, 2020-02-20:
OK thank you.
I will do that.
I already modified the gateway-config by adding Radioenge to the list.
I also added support for AU915 for RHF0M301, since the files are already in the folder and examples were just to make the call.

Sorry to be bothering you.
but Radioenge does not support the products it sells, so I'm having to do all the work for them kkk.
Good equipment, however without any support.

I'll check the Rasp image they sent to see if I can find the configuration file.

ffraporti, 2020-08-15:
@erdmanncross 

Acabei recebendo um gateway da RadioEnge essa semana, me deparei com o mesmo problema que você...

Não queria usar a imagem disponibilizada por eles, você conseguiu fazer funcionar com o chirpstack-concentratord? Pode me ajudar?

Grande abraço,
Felipe.

erdmanncross, 2020-08-15:
Então, dá pra usar com a configuração do risinrf (rhf3001 ou algo parecido)
só que nas últimas atualizações do concentratord tem que modificar uns
parâmetros nos arquivos, nada demais coisas de canal e tals

ffraporti, 2020-08-15:
Poxa.

Obrigado pela resposta rápida! Você sabe quais arquivos? Eu usei a
configuração do risinghf com reset PIN no 26, é isso mesmo?

Abraços,
Felipe

erdmanncross, 2020-08-15:
Então, no menu de configuração quando eu tava usando o concentratord, não
tinhas as opções pra usar AU915 então tinha que fazer na mão, e tbm estava
dando um erro pra usar GPS.
Eu deixei de usar na versão teste 5 se não me engano porque tinha que usar
umas libs e toda vez que mudava eu tinha que compilar tudo de novo.
Resolvi ir pra solução mais rápida e usar a imagem que veio no gateway e
instalando o Gateway bridge pra poder enviar direto por mqtt.
Eu estou usando um ds18b20 pra ler temp do case onde o Gateway está e
enviar pro application server, por isso não deu pra usar o concentratord.
Eu precisava que fosse rápido pois estava no final do mestrado e não podia
perder tempo. Mas ainda vou adicionar o suporte ao radioenge no
concentratord.

ffraporti, 2020-08-15:
Entendi.

Beleza! Muito obrigado pelas dicas e ajudas, vou dar uma olhada. Se
conseguir usar o gateway-os com radioenge tbm te aviso.

Grande abraço,
Felipe