chirpstack/chirpstack-concentratord #2: Is GPS required

mrbubble62, 2020-02-22:
I have gateway OS base running on pi Zero WiFi and forwarding to separate Chirpstack network server. The networkserver/application server is receiving join requests and sending join accept but they never get to the node.

```
Feb 22 19:24:47 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399487.930063245s
Feb 22 19:24:48 raspberrypi0-wifi user.info chirpstack-concentratord-sx1301[12451]: Frame received, uplink_id: 7a0dea0d-9b27-42e2-ba9a-a55b44dc0775, count_us: 1022705652, freq: 903700000, bw: 125000, mod: LoRa, dr: SF10
Feb 22 19:24:48 raspberrypi0-wifi user.info chirpstack-gateway-bridge[7841]: time="2020-02-22T19:24:48Z" level=info msg="backend/concentratord: uplink event received" uplink_id=7a0dea0d-9b27-42e2-ba9a-a55b44dc0775
Feb 22 19:24:48 raspberrypi0-wifi user.info chirpstack-gateway-bridge[7841]: time="2020-02-22T19:24:48Z" level=info msg="integration/mqtt: publishing event" event=up qos=0 topic=gateway/be8c8506f284c309/event/up uplink_id=7a0dea0d-9b27-42e2-ba9a-a55b44dc0775
Feb 22 19:24:48 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399488.930631287s
Feb 22 19:24:48 raspberrypi0-wifi user.info chirpstack-gateway-bridge[7841]: time="2020-02-22T19:24:48Z" level=info msg="integration/mqtt: downlink frame received" downlink_id=eba706b7-253b-4c9d-a11b-4b8aac47cdaf gateway_id=be8c8506f284c309
Feb 22 19:24:48 raspberrypi0-wifi user.info chirpstack-gateway-bridge[7841]: time="2020-02-22T19:24:48Z" level=info msg="backend/concentratord: forwarding downlink command" downlink_id=eba706b7-253b-4c9d-a11b-4b8aac47cdaf
Feb 22 19:24:49 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399489.931189329s
Feb 22 19:24:50 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399490.931945369s
Feb 22 19:24:51 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399491.93268741s
Feb 22 19:24:52 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399492.93343445s
Feb 22 19:24:53 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399493.934175491s
Feb 22 19:24:54 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399494.934981531s
Feb 22 19:24:55 raspberrypi0-wifi user.info chirpstack-concentratord-sx1301[12451]: Frame received, uplink_id: ee6f6bfc-1fdf-4b27-aa4f-6fc83118bbc2, count_us: 1029331459, freq: 903000000, bw: 500000, mod: LoRa, dr: SF8
Feb 22 19:24:55 raspberrypi0-wifi user.info chirpstack-gateway-bridge[7841]: time="2020-02-22T19:24:55Z" level=info msg="backend/concentratord: uplink event received" uplink_id=ee6f6bfc-1fdf-4b27-aa4f-6fc83118bbc2
Feb 22 19:24:55 raspberrypi0-wifi user.info chirpstack-gateway-bridge[7841]: time="2020-02-22T19:24:55Z" level=info msg="integration/mqtt: publishing event" event=up qos=0 topic=gateway/be8c8506f284c309/event/up uplink_id=ee6f6bfc-1fdf-4b27-aa4f-6fc83118bbc2
Feb 22 19:24:55 raspberrypi0-wifi user.info chirpstack-gateway-bridge[7841]: time="2020-02-22T19:24:55Z" level=info msg="integration/mqtt: downlink frame received" downlink_id=a20031f6-8ae9-4d7f-bb27-e3de24e0a43b gateway_id=be8c8506f284c309
Feb 22 19:24:55 raspberrypi0-wifi user.info chirpstack-gateway-bridge[7841]: time="2020-02-22T19:24:55Z" level=info msg="backend/concentratord: forwarding downlink command" downlink_id=a20031f6-8ae9-4d7f-bb27-e3de24e0a43b
Feb 22 19:24:55 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399495.935737571s
Feb 22 19:24:56 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399496.936311613s
Feb 22 19:24:57 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399497.937036653s
Feb 22 19:24:58 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399498.937775694s
Feb 22 19:24:59 raspberrypi0-wifi user.warn chirpstack-concentratord-sx1301[12451]: GPS time reference is not valid, age: 1582399499.938541734s

```
Node is running MCCI Cantenna compliance-otaa-halconfig.ino
```
6973425 (111574 ms): EV_TXSTART: ch=7 rps=0x04 (SF10 BW125 CR 4/5 Crc IH=0), datarate=0, opmode=88C, txend=6974038, avail=0
6973484 (111575 ms): +Tx LoRa, datum=0x17. opmode=88C
6996671 (111946 ms): radio_irq_handler_v2: LoRa, datum=0x8. opmode=88C
7308415 (116934 ms): EV_RXSTART: freq=927.5 rps=0x94 (SF10 BW500 CR 4/5 NoCrc IH=0), datarate=0, opmode=88C, txend=6996665, avail=0, rxtime=7309040, rxsyms=7
7309044 (116944 ms): +Rx LoRa Single, datum=0x0. opmode=88C
7309951 (116959 ms): radio_irq_handler_v2: LoRa, datum=0x80. opmode=88C
7370915 (117934 ms): EV_RXSTART: freq=923.3 rps=0x96 (SF12 BW500 CR 4/5 NoCrc IH=0), datarate=0, opmode=88C, txend=6996665, avail=0, rxtime=7371540, rxsyms=7
7371544 (117944 ms): +Rx LoRa Single, datum=0x0. opmode=88C
7375139 (118002 ms): radio_irq_handler_v2: LoRa, datum=0x80. opmode=88C
7375146 (118002 ms): EV_JOIN_TXCOMPLETE, saveIrqFlags 0x80, nLateRx=0 ticks=0
7376935 (118030 ms): EV_TXSTART: ch=64 rps=0x12 (SF8 BW500 CR 4/5 Crc IH=0), datarate=4, opmode=88C, txend=7377549, avail=0
7376995 (118031 ms): +Tx LoRa, datum=0x17. opmode=88C
7378781 (118060 ms): radio_irq_handler_v2: LoRa, datum=0x8. opmode=88C
7690525 (123048 ms): EV_RXSTART: freq=923.3 rps=0x91 (SF7 BW500 CR 4/5 NoCrc IH=0), datarate=4, opmode=88C, txend=7378775, avail=0, rxtime=7691150, rxsyms=14
7691154 (123058 ms): +Rx LoRa Single, datum=0x0. opmode=88C
7691389 (123062 ms): radio_irq_handler_v2: LoRa, datum=0x80. opmode=88C
7753025 (124048 ms): EV_RXSTART: freq=923.3 rps=0x96 (SF12 BW500 CR 4/5 NoCrc IH=0), datarate=4, opmode=88C, txend=7378775, avail=0, rxtime=7753650, rxsyms=7
7753654 (124058 ms): +Rx LoRa Single, datum=0x0. opmode=88C
7757249 (124115 ms): radio_irq_handler_v2: LoRa, datum=0x80. opmode=88C
7757257 (124116 ms): EV_JOIN_TXCOMPLETE, saveIrqFlags 0x80, nLateRx=0 ticks=0
```
Is my issue related to the missing GPS data?

mrbubble62, 2020-02-22:
Changing the model from "generic_us915_gps" to "generic_us915" still generates error messages about gps.
```
eb 22 19:53:10 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Disabling GPS mode for concentrator counter
Feb 22 19:53:10 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Current concentrator count_us: 62705041
Feb 22 19:53:10 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Concentrator drift, drift_us: -554
Feb 22 19:53:10 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Enabling GPS mode for concentrator counter
Feb 22 19:53:15 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Could not get GPS time, uplink_id: c085cded-5067-423a-a731-7a0c3ceafcf8, error: gps_ref_valid = false
Feb 22 19:53:15 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Could not get GPS epoch, uplink_id: c085cded-5067-423a-a731-7a0c3ceafcf8, error: gps_ref_valid = false
Feb 22 19:53:15 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Could not get GPS coordinates, uplink_id: c085cded-5067-423a-a731-7a0c3ceafcf8, error: gps_ref_valid = false
Feb 22 19:53:15 raspberrypi0-wifi user.info chirpstack-concentratord-sx1301[16368]: Frame received, uplink_id: c085cded-5067-423a-a731-7a0c3ceafcf8, count_us: 68110780, freq: 902500000, bw: 125000, mod: LoRa, dr: SF10
Feb 22 19:53:15 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:15Z" level=info msg="backend/concentratord: uplink event received" uplink_id=c085cded-5067-423a-a731-7a0c3ceafcf8
Feb 22 19:53:15 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:15Z" level=info msg="integration/mqtt: publishing event" event=up qos=0 topic=gateway/be8c8506f284c309/event/up uplink_id=c085cded-5067-423a-a731-7a0c3ceafcf8
Feb 22 19:53:16 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:16Z" level=info msg="integration/mqtt: downlink frame received" downlink_id=a092843b-5b20-4eb2-a273-1dea8592d9d9 gateway_id=be8c8506f284c309
Feb 22 19:53:16 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:16Z" level=info msg="backend/concentratord: forwarding downlink command" downlink_id=a092843b-5b20-4eb2-a273-1dea8592d9d9
Feb 22 19:53:22 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Could not get GPS time, uplink_id: 1615cea2-c0ad-4ad5-a1c9-31b35fba5047, error: gps_ref_valid = false
Feb 22 19:53:22 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Could not get GPS epoch, uplink_id: 1615cea2-c0ad-4ad5-a1c9-31b35fba5047, error: gps_ref_valid = false
Feb 22 19:53:22 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Could not get GPS coordinates, uplink_id: 1615cea2-c0ad-4ad5-a1c9-31b35fba5047, error: gps_ref_valid = false
Feb 22 19:53:22 raspberrypi0-wifi user.info chirpstack-concentratord-sx1301[16368]: Frame received, uplink_id: 1615cea2-c0ad-4ad5-a1c9-31b35fba5047, count_us: 74726139, freq: 903100000, bw: 125000, mod: LoRa, dr: SF7
Feb 22 19:53:22 raspberrypi0-wifi user.debug chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:22Z" level=debug msg="backend/concentratord: ignoring uplink event, CRC is not valid" crc_status=BAD_CRC uplink_id=1615cea2-c0ad-4ad5-a1c9-31b35fba5047
Feb 22 19:53:22 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Could not get GPS time, uplink_id: ff59b76b-a291-4a29-a6a5-57d01f4f7929, error: gps_ref_valid = false
Feb 22 19:53:22 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Could not get GPS epoch, uplink_id: ff59b76b-a291-4a29-a6a5-57d01f4f7929, error: gps_ref_valid = false
Feb 22 19:53:22 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Could not get GPS coordinates, uplink_id: ff59b76b-a291-4a29-a6a5-57d01f4f7929, error: gps_ref_valid = false
Feb 22 19:53:22 raspberrypi0-wifi user.info chirpstack-concentratord-sx1301[16368]: Frame received, uplink_id: ff59b76b-a291-4a29-a6a5-57d01f4f7929, count_us: 74882772, freq: 903000000, bw: 500000, mod: LoRa, dr: SF8
Feb 22 19:53:22 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:22Z" level=info msg="backend/concentratord: uplink event received" uplink_id=ff59b76b-a291-4a29-a6a5-57d01f4f7929
Feb 22 19:53:22 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:22Z" level=info msg="integration/mqtt: publishing event" event=up qos=0 topic=gateway/be8c8506f284c309/event/up uplink_id=ff59b76b-a291-4a29-a6a5-57d01f4f7929
Feb 22 19:53:22 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:22Z" level=info msg="integration/mqtt: downlink frame received" downlink_id=fe3cf9db-5eed-4c10-9da7-dfcc6ba1caad gateway_id=be8c8506f284c309
Feb 22 19:53:22 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:22Z" level=info msg="backend/concentratord: forwarding downlink command" downlink_id=fe3cf9db-5eed-4c10-9da7-dfcc6ba1caad
Feb 22 19:53:40 raspberrypi0-wifi user.debug chirpstack-concentratord-sx1301[16368]: Get gps coordinates error, error: gps_ref_valid = false
Feb 22 19:53:40 raspberrypi0-wifi user.info chirpstack-concentratord-sx1301[16368]: Publishing stats event, stats_id: f6804380-c834-4ee0-a969-9ff19b8bc216, rx_received: 3, rx_received_ok: 2, tx_received: 0, tx_emitted: 0
Feb 22 19:53:40 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:40Z" level=info msg="backend/concentratord: stats event received" stats_id=f6804380-c834-4ee0-a969-9ff19b8bc216
Feb 22 19:53:40 raspberrypi0-wifi user.info chirpstack-gateway-bridge[15911]: time="2020-02-22T19:53:40Z" level=info msg="integration/mqtt: publishing event" event=stats qos=0 stats_id=f6804380-c834-4ee0-a969-9ff19b8bc216 topic=gateway/be8c8506f284c309/event/stats
```

erdmanncross, 2020-02-22:
Which GPS module are you using?
I have this same problem using a Ublox-neo-6m at the gateway.
I tested it with another module as well, a ls20026 (SiRF Star III chipset).
when I check if the modules are receiving a valid location (sudo cat / dev / ttyAMA0), and everything is ok.
Only the concentrator does not recognize.

mrbubble62, 2020-02-23:
GPS module is Unicorecom UM220-III N it's embedded in the board I am using LoraGo PORT

mrbubble62, 2020-02-23:
More info...
GPS NMEA output is good until concentratord starts

> $GNRMC,153018.000,A,3855.179310,N,07702.392533,W,1.460,127.499,230220,,E,A*22
> 
> $GNGGA,153018.000,3855.179310,N,07702.392533,W,1,07,1.386,23.128,M,0,M,,*48
> 
> $GNGLL,3855.179310,N,07702.392533,W,153018.000,A,A*52
> 
> $GNGSA,A,3,32,31,25,26,22,14,16,,,,,,2.844,1.386,2.483*25
> 
> $GNGSA,A,3,,,,,,,,,,,,,2.844,1.386,2.483*27
> 
> $GPGSV,4,1,13,1,4,259,22,3,31,307,,4,5,301,,10,3,164,23*73
> 
> $GPGSV,4,2,13,14,68,78,27,16,20,195,15,22,44,281,17,23,2,302,*7F
> 
> $GPGSV,4,3,13,25,24,44,22,26,49,187,24,29,13,87,,31,72,353,22*78
> ...

Starting concentratord the data stream changes to 

> ▒▒▒▒▒▒▒▒▒▒▒... 

and in PuTTY log looks like..
> ð þ  ðþ  ð ð   À þ  À    þ  ð þþ  ð      þ  ð  ð À  ð ð  ðþ  ðð    ð   À þ  ð   À þ  ð  ð þ... 

A reboot will restore the GPS stream

brocaar, 2020-02-24:
> GPS module is Unicorecom UM220-III N

This module is not supported by the Semtech HAL: https://github.com/lora-net/lora_gateway#v500. The HAL uses the proprietary UBX output of the uBlox module to get the monotonic GPS epoch time, instead of the UTC timestamp.

mrbubble62, 2020-02-24:
So Sandbox Electronics LoraGo PORT is not supported, or GPS is not required?

erdmanncross, 2020-02-24:
by definition GPS is not necessary to operate the Gateway-OS.
i have a gateway that works normally without gps.
and even if the gps is not working the gateway continues to forward the msg normally to the NS

mrbubble62, 2020-02-24:
I will try to test with lora_gateway v4.1.3 which supported standard NMEA sentences.

mrbubble62, 2020-02-24:
My nodes are not receiving down link packets so join never completes. Just trying to eliminate GPS as a cause.  I have tried MCCI Arduino-lmic and ESP32-Paxcounter with similar results.

erdmanncross, 2020-02-24:
Sorry, are you using Chirpstack-Concentratord or Semtech as a forwarder?
can you post an image of the gateway page on your app server?
Which endDevice are you using?

mrbubble62, 2020-02-24:
#1 post shows logs from Chirpstack-Concentratord, end device is esp32+rfm95. I have created multiple end devices and applications, join requests are replied with join accept from NS, but end device never sees the reply.

brocaar, 2020-03-17:
I'm going to close this issue. The latest release of concentratord implements the `model_flags` which can be used to turn on / off `GNSS`.

As mentioned by @erdmanncross GPS / GNSS is not a requirement for a LoRa gateway. Only when you are going to use Class-B or geolocation, it is needed in order for precise timing.

The downlink not being received can be caused by many issues. Please do some searching on the forum: https://forum.chirpstack.io, you will find various solutions for various issues :)

closed 2020-03-17 by brocaar