chirpstack/chirpstack-concentratord #9: Poly packet forward feature

fernandobarreto, 2020-04-20:
- [X ] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## Summary

A new feature should be the poly packet forward in order to forward node messages to/from different network servers (chirpstack network server , TTN network server, etc...).

## What is the use-case?

This feature will enable TTN nodes and ChirpStack nodes to use the same gateway.

## Implementation description

I think the chirpstack-packet-multiplexer could be a guide for this or The Things Network Multi Protocol Packet Forwarder (https://github.com/kersing/packet_forwarder). This last one provides TTN packet forwarder and semtech packet forward (UDP). So, it can be used to provide packet forward to TTN and ChirpStack (semtech packet forward) at the same time. It will forward the messages to both network servers, but each one will process only messages from its node messages from its nodes, ignoring the other messages.

## Can you implement this by yourself and make a pull request?

Unfortunatly no...

brocaar, 2020-04-29:
The idea with the Concentratord is that you are able to run multiple packet-forwarder instances simultaniously. So instead having a poly-packet-forwarder, this makes it possible to have one TTN forwarder, one ChirpStack forwarder (the ChirpStack Gateway Bridge), ....

What would be good to have is a packet-forwarder on top of Concentratord which implements https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT. Then you can run as many instances as you like on top of Concentratord :)

brocaar, 2020-12-21:
This has been implemented by https://github.com/brocaar/chirpstack-udp-bridge :)

closed 2020-12-21 by brocaar