chirpstack/chirpstack-concentratord #11: Set reset pin for concentratord : IMST ic880A

kaboelliotpule, 2020-08-28:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## Summary

Allow “generic” configuration for IMST ic880A for setting the reset pin:
chirpstack-concentratord/chirpstack-concentratord-sx1301/src/config/vendor/imst/ic880a_eu868.rs

Currently the reset pin is fixed to pin 5 and this does not work as I need to set it to GPIO 25.

## What is the use-case?

To reset the pin on the concentrator board on a raspberry pi powered gateway

## Implementation description

Maybe change :
reset_pin: Some(5) -> reset_pin: Some(conf.gateway.reset_pin)

## Can you implement this by yourself and make a pull request?
No

proffalken, 2020-10-05:
I'd propose that the config file overrides the default as opposed to the current situation which appears to be the other way around:

```
# Reset pin.
#
# Note: most model configurations come with a pre-defined reset_pin, in which
# case the setting below will be ignored.
reset_pin=22
```

If using the board from https://www.tindie.com/products/gnz/imst-ic880a-lorawan-backplane/ then the reset pin is pin 22, not pin 5, so being able to leave 5 as the default and then override it in the config if necessary would be great!

proffalken, 2020-10-05:
Looks like it's https://github.com/brocaar/chirpstack-concentratord/blob/master/chirpstack-concentratord-sx1301/src/config/vendor/imst/ic880a_eu868.rs#L149 where the change needs to be made, however I'm not familiar enough with Rust to make this change.

brocaar, 2020-10-05:
@kaboelliotpule @proffalken I believe the above comment should fix the issue :)

closed 2020-10-05 by brocaar

proffalken, 2020-10-06:
And this is why I continue to base a business on Chirpstack, thanks @brocaar - I'll look forward to the next release!

kaboelliotpule, 2020-10-09:
@brocaar When will the next release come out incorporating the above change?

SKRahulSK, 2020-10-13:
@brocaar 
Could you please publish a new release with this change?

brocaar, 2020-10-14:
@proffalken @kaboelliotpule @SKRahulSK I will issue a new release asap. I have a couple other things that I must work on first. Without any obstacles, this probably will be early next week.

proffalken, 2020-10-14:
Amazing, thanks @brocaar, really appreciate it!

kaboelliotpule, 2020-10-14:
@brocaar, much appreciated. Thanks

SKRahulSK, 2020-10-16:
@brocaar Thank you 👍

brocaar, 2020-10-16:
Unfortunately there is a small blocker until I can release a new ChirpStack Gateway OS version with this version: https://github.com/openembedded/meta-openembedded/issues/285.

It looks like it is known what caused the issue and a fix is underway :)

fe64970103, 2020-12-29:
Hello. As I understand the fix for this wasn't yet released. Is there a workaround for this in the meantime? Thanks, Marko.

brocaar, 2020-12-29:
Correct, I have had a few issues updating the ChirpStack Gateway OS layers. I'm currently building new images for doing some final testing. I do hope to release a new GW OS version later this week.

fe64970103, 2020-12-29:
Thanks for the fast response.  Appreciated!

fe64970103, 2020-12-29:
From what is worth I've build the standalone concentratord from git and deployed it on my target PI. It seems to be working. At least I can see it getting frames and outputting messages like:

Dec 29 18:54:53 raspberrypi3 user.info chirpstack-gateway-bridge[433]: time="2020-12-29T18:54:53Z" level=info msg="backend/concentratord: stats event received" stats_id=161f0e0a-d9f8-47d8-bb08-a812c426e555
Dec 29 18:54:53 raspberrypi3 user.info chirpstack-gateway-bridge[433]: time="2020-12-29T18:54:53Z" level=info msg="integration/mqtt: publishing event" event=stats qos=0 stats_id=161f0e0a-d9f8-47d8-bb08-a812c426e555 topic=gateway/b827ebfffe18a588/event/stats
Dec 29 18:54:53 raspberrypi3 user.info chirpstack-network-server[960]: time="2020-12-29T18:54:53Z" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=b827ebfffe18a588 stats_id=161f0e0a-d9f8-47d8-bb08-a812c426e555
Dec 29 18:54:53 raspberrypi3 user.warn chirpstack-network-server[960]: time="2020-12-29T18:54:53Z" level=warning msg="gateway/stats: stats received by unknown gateway" ctx_id=161f0e0a-d9f8-47d8-bb08-a812c426e555 gateway_id=b827ebfffe18a588

brocaar, 2021-01-06:
I have just released a new Gateway OS version :)

kaboelliotpule, 2021-01-07:
Awesome, thanks Orne

proffalken, 2021-01-07:
Woohoo! Thanks @brocaar !

fe64970103, 2021-01-07:
Thanks Orne. Trying it out today.

commit fee1bed2 (2020-10-05): Implement overwriting reset pin for RPi shields.

When the reset_pin is set to 0 / unconfigured, the reset_pin value from
the vendor configuration is used, else the reset_pin from the
configuration file is used.

Closes #11.