chirpstack/chirpstack-concentratord #13: GPS time reference is not valid (Multitech)

chopmann, 2020-10-19:
- [x ] The issue is present in the latest release.
- [x ] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## What happened?
Build and installed concentratord for Multitech. Messages are being sent okay, GPS is not working thou. I checked the system time and date gives an accurate date. Gateway has a clear sky view and used to send a fix while running with the pkt-fwd.

## What did you expect?
GPS to work.
## Steps to reproduce this issue

Steps:

1. Install https://static.iot.wolfsburg.digital/chirpstack/chirpstack-concentratord_3.0.2-1-gfee1bed-r1_arm926ejste.ipk
2. Start concentratord
3. Check /var/log/messages

## Could you share your log output?

```shell
PRODWOBMTCDT09:/home/mtadm# /etc/init.d/chirpstack-concentratord-ap1 restart
stopping chirpstack-concentratord
Found MTAC-LORA-H-868 with MTAC-LORA-1.5 hardware at ap1
Starting chirpstack-concentratord
PRODWOBMTCDT09:/home/mtadm# tail /var/log/messages -f                       
Oct 19 09:41:20 PRODWOBMTCDT09 user.info chirpstack-concentratord-sx1301[29766]: Starting the concentrator
Oct 19 09:41:23 PRODWOBMTCDT09 user.info chirpstack-concentratord-sx1301[29766]: Creating socket for publishing events, bind: ipc:///tmp/concentratord_event
Oct 19 09:41:23 PRODWOBMTCDT09 user.info chirpstack-concentratord-sx1301[29766]: Creating socket for receiving commands, bind: ipc:///tmp/concentratord_command
Oct 19 09:41:23 PRODWOBMTCDT09 user.info chirpstack-concentratord-sx1301[29766]: Initializing JIT queue, capacity: 32
Oct 19 09:41:23 PRODWOBMTCDT09 user.info chirpstack-concentratord-sx1301[29766]: Starting GPS validation loop
Oct 19 09:41:23 PRODWOBMTCDT09 user.info chirpstack-concentratord-sx1301[29766]: GPS TTY port opened for GPS synchronization, gps_tty_path: /dev/ttyXRUSB2
Oct 19 09:41:24 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603093284.954873654s
Oct 19 09:41:25 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603093285.955473528s
Oct 19 09:41:26 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603093286.95601952s
Oct 19 09:41:27 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603093287.956599773s
Oct 19 09:41:28 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603093288.957125485s
Oct 19 09:41:29 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603093289.957631936s
Oct 19 09:41:30 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603093290.958167487s
Oct 19 09:41:31 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603093291.958702257s
Oct 19 09:41:32 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603093292.959222687s
Oct 19 09:41:33 PRODWOBMTCDT09 user.info chirpstack-concentratord-sx1301[29766]: Frame received, uplink_id: a5d90fd3-d464-47f0-bcd4-fc19ac7778e6, count_us: 11925628, freq: 868300000, bw: 125000, mod: LoRa, dr: SF12
```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.12.2 |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |     3.0.2-1-gfee1bed-r1    |

brocaar, 2020-10-19:
Please note that it takes couple of seconds to get the correct time reference. I see this warning too, but it disappears shortly after. From your logs, I see that the concentratord has been started on 10 seconds ago.

chopmann, 2020-10-19:
I know i know, checked before that and the gateway was running for 2 days and the application-server was reporting our "default" location when not deployed. 
`Oct 19 11:43:04 PRODWOBMTCDT09 user.warn chirpstack-concentratord-sx1301[29766]: GPS time reference is not valid, age: 1603100584.576004105s`
checked again a few minutes ago.

chopmann, 2020-10-19:
I'll keep the Issue open, but seems to be just this GW. Another (same model and config) is running okay and the logs look good.

brocaar, 2020-10-19:
Could you confirm that the path to the GNSS device is `/dev/ttyXRUSB2`? That is embedded within the Concentratord config profile for the Conduit (what I took from my Conduit).

chopmann, 2020-10-19:
The path is okay. I see the ublox-binary if I cat it.
```
PRODWOBMTCDT09:~$ sudo -s
PRODWOBMTCDT09:/home/mtadm# cat /dev/ttyXRUSB2                     
BJJ
sXFRNRB	BBBCBBB~)jHFB%bIh#FJJBJFhHBGGG^KOBNNFNn@BNNHBBSnȀBLnjFlnjnXF^C
PRODWOBMTCDT09:/home/mtadm# 
```

chopmann, 2020-10-20:
Found out the Issue on two gateways.
In our case was gpsd was installed in one of the gateways and in the other an old packet-fwd process was still alive.
Both procs access the gps and block each other - depending on boot. After we killed and stopped the procs did 
```shell
cd /sys/devices/platform/mts-io
echo 0 >gnss-reset; sleep 1 ; echo 1 >gnss-reset
```
restarted both concentratord and gw-bridge, the GW was reporting the correct gps position.

closed 2020-10-20 by chopmann