chirpstack/chirpstack-concentratord #27: Add SW1302 LoraWAN Gateway HAT for ChirpStack OS Gateway and Raspberry PI 4B

dyane6, 2022-08-13:
I have a project to manage water metering sensors based on LoraWAN technology for my residence.
I would like to create a LoraWAN Gateway based on ChirpStack OS Gateway, Raspberry PI 4B board and SX1302 LoRaWAN Gateway hat (www.waveshare.com). 
I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.
It could be a new hardware managed by the ChirpStack OS Gateway and easily integrated in the software 
https://www.waveshare.com/wiki/SX1302_LoRaWAN_Gateway_HAT

I don't know how implement and introduce by myself this hat. I need help

errolt, 2022-11-24:
I have been able to select the Waveshare hat during OS setup. Only problem I currently see is that the GPS isn't read.

dyane6, 2022-11-24:
Hello, Orne recently added the waveshare hat SX1302 in the gateway OS. But the software doesn't work on my Raspberry PI 4B (look at the picture). I suspect a CM4 issue. Which board do you use to make th test?
[image]

errolt, 2022-11-24:
I used both Pi3B and Pi4B.

I have never seen that output before, but I run all my Pis headless, so it might be there and never noticed.

dyane6, 2022-11-24:
You are lucky! The issue is known. I installed a test version from Orne built in last January and the software works but the hat Waveshare SX1302 is not integrated.
https://github.com/chirpstack/chirpstack-gateway-os/issues/81
With or without the hat, the issue is present.

errolt, 2022-11-24:
Is your bootloader for the pi4 up to date? I cant remember how, but booting from a raspbian os, there is a way to update the bootloader located in flash on the pi4 pcb.

errolt, 2022-11-24:
Just looked, my bootloader isn't up to date either:
```
$ sudo rpi-eeprom-update 
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Thu 29 Apr 2021 16:11:25 UTC (1619712685)
    LATEST: Tue 26 Apr 2022 10:24:28 UTC (1650968668)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
```

dyane6, 2022-11-24:
I have as the same bootloader version as you. See below.
BOOTLOADER: up to date
   CURRENT: Fri  4 Nov 15:19:15 UTC 2022 (1667575155)
    LATEST: Fri  4 Nov 15:19:15 UTC 2022 (1667575155)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

errolt, 2022-11-24:
Don't know if this is the issue, but no, my bootloader is old:
`   CURRENT: Thu 29 Apr 2021 16:11:25 UTC (1619712685)
`
Your bootloader is latest:
` CURRENT: Fri 4 Nov 15:19:15 UTC 2022 (1667575155)`

I'll update to latest bootloader, when I get a chance, to see if GatewayOS still boots.

dyane6, 2022-11-25:
My Raspberry board is Model 4B V1.5, revision do3115 Soc BCM2711, 8GB RAM. And you?

errolt, 2022-11-25:
Dont know what versions. One Pi4B board is 1GB ram, the other is 4GB ram. Both boot fine.

Just tested, neither of my 8GB Pi4's will boot GatewayOS.

errolt, 2022-12-18:
The issue seems to be that 8GB Pi4s doesn't work. I suspect that the OS is 32bit based, and crashes when more than 4GB ram is available.

dyane6, 2022-12-18:
Good clue to explain the crash. 
The new version of gateway OS 4.0.2 boots normally on my Pi4 8GB but crashes when the command sudo gateway-config is done. See the screenshot.

[image]

brocaar, 2022-12-19:
@dyane6 have you tried to re-flash the sdcard to make sure you started with a clean state?

brocaar, 2022-12-19:
> The issue seems to be that 8GB Pi4s doesn't work. I suspect that the OS is 32bit based, and crashes when more than 4GB ram is available.

It should not. The Raspberry Pi OS 32bit version can be used for all Pi versions. I believe the full 8GB will be available when using the 32bit OS, but the maximum memory per process is limited to 3GB.

The boot issue could have been related to a different issue (Node-RED / NodeJS crashing). I believe there was a bug in some of the cross-compiling scripts which might have generated some invalid CPU instructions. The latest GW OS version rolled back some of these recipe updates.

brocaar, 2022-12-19:
You will see that screen until the PostgreSQL database has been initialized. I'm not sure if that is considered a crash? Depending the speed of your sd card, this could take a few minutes.

I just reflashed a sd card with the v4.0.2 full image, and this "The database is initializing..." goes away after about a minute or so.

closed 2022-12-19 by brocaar

dyane6, 2022-12-19:
@brocaar
I agree to close the topic because the initial issue is solved. I gonna re-flash the SD card 2 or 3 times hoping postgresql_dbs_created file will be created.