chirpstack/chirpstack-concentratord #29: Provide an option to forward packets with bad/missing CRC

mullerch, 2022-08-24:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## Summary

Add an option, in the configuration file, to forward packets with bad/missing CRC. The same way the Semtech UDP Packet Forwarder does.

## What is the use-case?

In case of device development, testing, debugging and diagnostic those packets can be useful on the server side.

## Implementation description

At the check if CRC is ok, add a check on the options to see if the packet must be dropped.

Also, it seems that there is a little "bug" actually. If the CRC is missing, the log will report that there was a packet with bad crc: https://github.com/chirpstack/chirpstack-concentratord/blob/039b0b0058639f6b9365dcd60e0f0dfd9ec261e4/chirpstack-concentratord-2g4/src/handler/uplink.rs#L27 

WARNING: the UDP forwarder connecting to the concentratord expects the CRC to be OK at this time and [does override status](https://github.com/chirpstack/chirpstack-udp-forwarder/blob/master/src/structs.rs#L250). This means there must first be an adaptation of this software.

## Can you implement this by yourself and make a pull request?

Yes possibly.

brocaar, 2022-08-26:
Hi @mullerch I removed this (also from the v4 Protobuf messages), but I don't mind adding this back if there is an use-case for this.

brocaar, 2023-01-16:
This is implemented by https://github.com/chirpstack/chirpstack-concentratord/commit/b9c36dd01e9b61b336e33f22afe6540e630624d1.

closed 2023-01-16 by brocaar

brocaar, 2023-01-16:
This implements the filtering for the UDP forwarder: https://github.com/chirpstack/chirpstack-udp-forwarder/commit/6a7fb8ac93bd301ee9a72e2637bd572d95b599b6.