chirpstack/chirpstack-concentratord #49: Add fake RX time option

mullerch, 2023-02-16:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## Summary

Add an option to set the RX time to the system time if there is no time set from the GPS.

## What is the use-case?

This option exists in the Gateway Bridge's Semtech UDP backend. Adding this feature to the concentratord would make the Concentratord + CSGB a replacement for Semtech UDP Packet Forwarder + CSGB when this feature is needed.

## Implementation description

Technically speaking it could be implemented in the CSGB instead, but I think this is more hardware related and therefore belong to the concentratord.

## Can you implement this by yourself and make a pull request?

I don't know where it belongs and the implication of the implementation of this for the different HAL implementations.

brocaar, 2023-06-05:
Hi @mullerch, sorry for the delay. I had quite a full backlog...

I agree this is useful and I'm looking at this right now. What do you think if the Concentratord would always fallback on system-time in case there is no GPS (fix)?

We have the `rx_info.time` and `rx_info.time_since_gps_epoch`. Currently, I believe both fields are empty in case no GPS time is available and both are set if there is a GPS time available.

What if we always set the `time` field regardless of GPS time availability?

mullerch, 2023-06-05:
HI @brocaar, thanks for taking some time for this!

I think, always having the time is a good idea. We just need to make sure the protocol defines which data comes from which source in which scenario to avoid confusion (or maybe have a time_source field?).

The only downside of this change is usage of more bandwidth if forwarded as is. However, I think this should be solved in the UDP/MQTT forwarder if needed as it's network related.

commit 879059a5 (2023-06-06): Implement time fallback option.

If enabled, the system-time of the gateway will be used if the gateway
does not have a GPS time source.

Closes #49.
closed 2023-06-06 by brocaar