chirpstack/chirpstack-concentratord #65: Chirpstack-consertratord-sx1302 backend - Fine Timestamp is missing from uplink event?

LouneCode, 2023-06-11:
- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## Summary

I have investigated the consentratord backend code and found that the Fine timestamp information basis on PPS signal missing from current implementation?

Consertratord-sx1302 implementation has been build on Semtechs's HAL library (libloragw). Sx1302_hal library includes code for Fine timestamp implementation (for hardware with PPS signal). It seems that the Fine Timestamp solution can be added to consertratord-sx1302 backend quite easily. 

Similar cases can be found from following issues: 
* https://github.com/chirpstack/chirpstack-gateway-bridge/issues/228
* https://github.com/lorabasics/basicstation/issues/177

&nbsp;

## What is the use-case?
A Fine Timestamp is used to get the TDoA geolocation of the transmitter (sensor).

## Implementation description  -  Solution proposal

Add Fine Timestamp information to the up-event of the concentratord-sx1302 backend.

&nbsp;

## 1) consertratord.toml

consertratord.toml file found in the same folder where `chirpstack-concentratord-sx1302` executable is. Following lines to be added into consertratord.toml configuration file for enabling Fine Timestamp functionality: 
* `model_flags=["USB","GNSS"]`
* `[gateway.fine_timestamp]` 
* `enable=true`
* `mode="ALL_SF"`

&nbsp;

```
# LoRa gateway configuration.
[gateway]

  # Antenna gain (dB).
  antenna_gain=0

  # Public LoRaWAN network.
  lorawan_public=true

  # Gateway vendor / model.
  #
  # This configures various vendor and model specific settings like the min / max
  # frequency and TX gain table.
  #model="rak_2287_eu868"
  model="semtech_sx1302css868gw1_eu868"

  # Gateway vendor / model flags.
  #
  # Flag can be used to configure additional vendor / model features. The
  # following flags can be used:
  #
  #   Global flags:
  #     GNSS - Enable GNSS / GPS support
  #     USB  - Use USB for concentrator communication (default is SPI)
  model_flags=["USB","GNSS"]

  # Time fallback.
  #
  # In case the gateway does not have a GNSS module or is unable to aquire a
  # GNSS fix, use the system-time for setting the 'time' field on RX.
  time_fallback_enabled=true

  #Fine Timestamp
  #mode: HIGH_CAPACITY or ALL_SF
  [gateway.fine_timestamp]
    enable=true
[... 168 lines cut ...]
}

```

&nbsp;

I wrote a simple Rust zeromq client program for printing up-events of the consertratord-sx1302 backend. Now Fine Timestamp information found in `fine_time_since_gps_epoch: Some(Duration { seconds: 0, nanos: 48290698 })` Structure as follows. In this example second part is allway zere. Only nano second part matters for the geolocation calculations.    

```

Topic: up
phy_payload: [64, 97, 158, 199, 1, 128, 245, 148, 3, 132, 29, 75, 233, 12, 129, 60, 199, 85, 34, 198, 216, 151, 189]
phy_payload: UplinkRxInfo { gateway_id: "0016c001ff1f11ddd", uplink_id: 1836980030, time: Some(Timestamp { seconds: 1686508758, nanos: 116930490 }), time_since_gps_epoch: None, fine_time_since_gps_epoch: Some(Duration { seconds: 0, nanos: 48290698 }), rssi: -99, snr: 12.25, channel: 3, rf_chain: 0, board: 0, antenna: 0, location: None, context: [57, 90, 153, 82], metadata: {}, crc_status: CrcOk }
Topic: up
phy_payload: [64, 97, 158, 199, 1, 128, 246, 148, 3, 86, 239, 135, 219, 156, 240, 225, 92, 184, 205, 19, 254, 61, 94]
phy_payload: UplinkRxInfo { gateway_id: "0016c001ff1f1ddd", uplink_id: 2518488021, time: Some(Timestamp { seconds: 1686508788, nanos: 161273198 }), time_since_gps_epoch: None, fine_time_since_gps_epoch: Some(Duration { seconds: 0, nanos: 91829739 }), rssi: -42, snr: 11.25, channel: 1, rf_chain: 1, board: 0, antenna: 0, location: None, context: [59, 37, 6, 226], metadata: {}, crc_status: CrcOk }
Topic: up
phy_payload: [64, 97, 158, 199, 1, 128, 247, 148, 3, 167, 7, 251, 70, 240, 160, 220, 53, 167, 110, 184, 203, 240, 19]
phy_payload: UplinkRxInfo { gateway_id: "0016c001ff1f1ddd", uplink_id: 1110056337, time: Some(Timestamp { seconds: 1686508818, nanos: 194757382 }), time_since_gps_epoch: None, fine_time_since_gps_epoch: Some(Duration { seconds: 0, nanos: 134425134 }), rssi: -42, snr: 9.5, channel: 2, rf_chain: 1, board: 0, antenna: 0, location: None, context: [60, 239, 112, 194], metadata: {}, crc_status: CrcOk }

```

## Can you implement this by yourself and make a pull request?

Yes ! can. I have a smoke tested this solution proposal and it seems to work .  But first let's wait for the review  and comments of the idea....

brocaar, 2023-06-12:
Hi @LouneCode please do create a pull-request for this :+1: 

I think we can use the seconds part of the `time_since_gps_epoch` for the `fine_time_since_gps_epoch` + the `packet.ftime` as `nanos`, then we have the full duration since GPS epoch

LouneCode, 2023-06-12:
OK, I'll try make a pull-request with the suggested improvements.

commit d3a3c7bb (2023-06-13): Add fine-timestamp to concentratord-sx1302.  (#66)

Closes #65.

---------
closed 2023-06-13 by brocaar