chirpstack/chirpstack-concentratord #85: Seeed WM1302 downlink message issue

kartsan, 2023-08-04:
- [X] The issue is present in the latest release.
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## What happened?

When GPS is in use and after 1-2 hours, the devices stop getting ACKs from the gateway and they start sending Join messages.  However, they cannot Join after that. According to the Chirpstack UI, ACKs and Join Accepts are sent but the devices are not satisfied.

## What did you expect?

With GPS in use, the devices continue sending messages and receive ACKs from the gateway.

## Steps to reproduce this issue

Steps:

1. Use Seeed WM1302 card on Pi Hat with Raspberry PI and install the latest Chirpstack OS on it.
2. Connect LoRaWAN antenna on WM1302 card AND GPS module/antenna on the Pi Hat.
3. Configure WM1302 card in use with gateway-config via LoRa concentrator shield.
4. Have LoRaWAN devices in your network with uplink acknowledge policy set on.
5. Wait ~2 hours with 3 minutes message interval on the LoRaWAN devices.

On the other hand, if I remove the GPS antenna from Pi Hat connector and go through the previously mentioned steps, I don't  see this issue. Instead I get the following warning every second (example):

```
Aug 04 09:27:43 raspberrypi4 user.warn chirpstack-concentratord-sx1302[538]: GPS time reference is not valid, age: 1691141263.592160279s
```

## Could you share your log output?

I don't see any change in the log messages between working and non-working (ACKs not received) state. When GPS is connected, I do see this message every two minutes:

```
Aug 03 21:02:34 raspberrypi4 user.warn chirpstack-concentratord-sx1302[537]: Enqueue beacon failed, error: TooEarly
```

Let me know if other logs are required.

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Chirpstack OS  | v4.3.0 |
| Concentratord       | v4.1.1 and v4.2.0  |

brocaar, 2023-08-04:
Do you mean ChirpStack Gateway OS v4.2.0-test.1?

If that is the case, then I might have found the issue. I was initially puzzled, because I was getting very strange GPS times from my GNSS module (different shield, but that is not relevant in this case), causing the Concentratord to eventually crash after some time. It noticed that Rust `bindgen` was using the AMD64 sysroot of the build environment for generating ARMv7 bindings. This was causing memory mis-alignments in the Rust to Semtech HAL "glue" causing the GPS time to look random.

This has been fixed and I will build a new test-image next week.

---

For reference, this was the fix (adding `--sysroot=$(TOOLCHAIN_DIR)`):

https://github.com/chirpstack/chirpstack-openwrt-feed/blob/master/chirpstack/chirpstack-concentratord/Makefile#L21

kartsan, 2023-08-04:
> Do you mean ChirpStack Gateway OS v4.2.0-test.1?

No I tried released version 4.2.0 of Concentrator binary on OS version 4.1.1 just to test if this issue has been fixed. The results were the same.

brocaar, 2023-09-11:
I'm closing this issue as this commit probably resolves this issue:
https://github.com/chirpstack/chirpstack-concentratord/commit/ceca631a6540e77fc518ff4e988078c3d42294fc

The Seeed GNSS module is not supported by the Semtech HAL. This was already released as v4.2.2 ([v4.2.2](https://www.chirpstack.io/docs/chirpstack-concentratord/changelog.html#v422)).

Please let me know if it doesn't fix your issue.

closed 2023-09-11 by brocaar

kartsan, 2023-09-11:
I'm currently testing with version 4.2.3 and I have not experienced this issue. Thank you!