chirpstack/chirpstack-concentratord #92: Waveshare 1302 Lorawan Gateway Hat - AU915 Support

tvalles, 2023-08-30:
- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## Summary

Support for Waveshare 1302 Lorawan Gteway Hat - AU915 Support

tvalles, 2023-09-11:
> I'm not sure if this module is intended to work on the 915 MHz frequency:
>
> [image: image]
> <https://user-images.githubusercontent.com/165497/267045467-57c4e324-9247-4d09-be02-b0766d3cd4ff.png>
>
> https://www.waveshare.com/wiki/SX1302_LoRaWAN_Gateway_HAT

Tks for the response, but the module i have is this with the 915Mhz.

[image: image.png]

brocaar, 2023-09-11:
I think images do not work when replying by e-mail. All I can see is `[image: image.png]`

tvalles, 2023-09-11:
> I think images do not work when replying by e-mail. All I can see is [image:
> image.png]

Ok, this link is the same i have.
https://www.amazon.com.br/Waveshare-Compatible-Raspberry-Transmission-Multi-Band/dp/B0BGQ165KT

tvalles, 2023-09-11:
> Thanks, I guess then their wiki is a outdated, as it does not mention a
> US915 / AU915 model:
> https://www.waveshare.com/wiki/SX1302_LoRaWAN_Gateway_HAT
>
> I'll add the US915 / AU915 regions soon.

Sounds good. Thanks.

tvalles, 2023-11-09:
Hi, its implemented yet. I installed the latest version and only supports EU Region.

brocaar, 2023-11-10:
It is implemented in this version:

https://www.chirpstack.io/docs/chirpstack-concentratord/changelog.html#v433

Using the `region` configuration in the TOML config you can configure one of the supported regions.

commit a1289677 (2023-10-11): sx1302: Add missing Waveshare SX1302 Gateway HAT regions.

Fixes #92.