chirpstack/chirpstack-concentratord #107: Chripstack-concentratord-sx1302 v4.3.0 crashed with Segmentation fault

LouneCode, 2023-10-04:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## Chripstack-concentratord-sx1302 v4.3.0 crashed with segmentation fault when using fine timestamping.
## Test environment
- Test environment: Ubuntu 22.04.3 LTS (GNU/Linux 5.15.0-79-generic x86_64) 
- Dev environment and code build as described on this repository.
- Cross build target: x86_64-unknown-linux-musl
- Module: Chirpstack-consertratord-sx1302
- Lora HW: Semtech sx1303css868gw with a PPS signal 

The Chripstack-concentratord-sx1302 module is compiled according to the instructions in this repository. 

## Consentratord crashed randomly ... 
Consentratord crashed randomly with SIGSEGV signal Segmentation fault. It seems that segmentation fault occurs (only?) if the Fine Stamping is enabled.  

Fine timestamping is enabled in a concentratord configuration file as follows In the test case: 

```
sx1303.toml
...

  # Fine_timestamp
  # enable
  #   true
  #   false
  # mode
  #   "ALL_SF"
  #   "HIGH_CAPACITY"
[gateway.fine_timestamp]
  enable=true
  mode="ALL_SF"
...
```

Consentratord start command: `./chirpstack-concentratord-sx1302 -c sx1303.toml`

Concentratord runs for a random amount of the time and crashes with a segmentation fault error. The log does not show the reason or any information about the crash. Only hint is that segmentation fault has been raised. Not good.

I've investigated and debugged chirpstack-consentratord-sx1302 code and found that the problem lies behind the Rust FFI in the customized [c code of sx1302_hal layer](https://github.com/brocaar/sx1302_hal). 

After a long run I got a full back trace log with the gdb debugger. 

```
#0  0x00007ffff772224a in compare_pkt_tmst ()
No symbol table info available.
#1  0x00007ffff7c01d0f in sift () at ../src_musl/src/stdlib/qsort.c:103
No locals.
#2  0x00007ffff7c01eb1 in trinkle () at ../src_musl/src/stdlib/qsort.c:154
No locals.
#3  0x00007ffff7c02082 in __qsort_r () at ../src_musl/src/stdlib/qsort.c:200
No locals.
#4  0x00007ffff7723d2a in lgw_receive ()
No symbol table info available.
#5  0x00007ffff771c5e7 in libloragw_sx1302::hal::receive () at libloragw-sx1302/src/hal.rs:932
        _guard = std::sync::mutex::MutexGuard<()> {lock: 0x7ffff7ff8c88 <<libloragw_sx1302::mutex::CONCENTATOR as core::ops::deref::Deref>::deref::__stability::LAZY+4>, poison: std::sync::poison::Guard {panicking: false}}
        packets = [
          libloragw_sx1302::wrapper::lgw_pkt_rx_s {freq_hz: 868500000, freq_offset: 854, if_chain: 2, status: 16, count_us: 254499580, rf_chain: 1, modem_id: 9, modulation: 16, bandwidth: 4, datarate: 7, coderate: 1, rssic: -40.3999939, rssis: -41.3999939, snr: 9.5, snr_min: 0, snr_max: 0, crc: 57943, size: 23, payload: [64, 38, 51, 248, 0, 128, 148, 143, 3, 200, 183, 189, 141, 158, 16, 176, 97, 176, 164, 99, 92, 183, 23, 0 <repeats 233 times>], ftime_received: true, ftime: 877986163},
          libloragw_sx1302::wrapper::lgw_pkt_rx_s {freq_hz: 868100000, freq_offset: -24138, if_chain: 0, status: 17, count_us: 254499587, rf_chain: 1, modem_id: 11, modulation: 16, bandwidth: 4, datarate: 7, coderate: 1, rssic: -96.3999939, rssis: -100.399994, snr: -7.25, snr_min: 0, snr_max: 0, crc: 57951, size: 23, payload: [112, 38, 55, 232, 2, 12, 148, 143, 195, 240, 183, 190, 193, 158, 17, 118, 97, 128, 164, 163, 124, 183, 177, 0 <repeats 233 times>], ftime_received: false, ftime: 0},
          libloragw_sx1302::wrapper::lgw_pkt_rx_s {freq_hz: 868100000, freq_offset: -24138, if_chain: 0, status: 17, count_us: 254499588, rf_chain: 1, modem_id: 10, modulation: 16, bandwidth: 4, datarate: 7, coderate: 1, rssic: -96.3999939, rssis: -100.399994, snr: -7.5, snr_min: 0, snr_max: 0, crc: 57942, size: 23, payload: [112, 68, 183, 248, 3, 8, 144, 143, 195, 112, 183, 238, 4, 158, 19, 118, 97, 128, 164, 163, 92, 183, 129, 0 <repeats 233 times>], ftime_received: false, ftime: 0},
          libloragw_sx1302::wrapper::lgw_pkt_rx_s {freq_hz: 868500000, freq_offset: 854, if_chain: 2, status: 16, count_us: 254499580, rf_chain: 1, modem_id: 8, modulation: 16, bandwidth: 4, datarate: 7, coderate: 1, rssic: -40.3999939, rssis: -41.3999939, snr: 11.5, snr_min: 0, snr_max: 0, crc: 57943, size: 23, payload: [64, 38, 51, 248, 0, 128, 148, 143, 3, 200, 183, 189, 141, 158, 16, 176, 97, 176, 164, 99, 92, 183, 23, 0 <repeats 233 times>], ftime_received: false, ftime: 0},
          libloragw_sx1302::wrapper::lgw_pkt_rx_s {freq_hz: 0, freq_offset: 0, if_chain: 0, status: 0, count_us: 0, rf_chain: 0, modem_id: 0, modulation: 0, bandwidth: 0, datarate: 0, coderate: 0, rssic: 0, rssis: 0, snr: 0, snr_min: 0, snr_max: 0, crc: 0, size: 0, payload: [
              0 <repeats 256 times>], ftime_received: false, ftime: 0},
          libloragw_sx1302::wrapper::lgw_pkt_rx_s {freq_hz: 0, freq_offset: 0, if_chain: 0, status: 0, count_us: 0, rf_chain: 0, modem_id: 0, modulation: 0, bandwidth: 0, datarate: 0, coderate: 0, rssic: 0, rssis: 0, snr: 0, snr_min: 0, snr_max: 0, crc: 0, size: 0, payload: [
              0 <repeats 256 times>], ftime_received: false, ftime: 0},
          libloragw_sx1302::wrapper::lgw_pkt_rx_s {freq_hz: 0, freq_offset: 0, if_chain: 0, status: 0, count_us: 0, rf_chain: 0, modem_id: 0, modulation: 0, bandwidth: 0, datarate: 0, coderate: 0, rssic: 0, rssis: 0, snr: 0, snr_min: 0, snr_max: 0, crc: 0, size: 0, payload: [
              0 <repeats 256 times>], ftime_received: false, ftime: 0},
          libloragw_sx1302::wrapper::lgw_pkt_rx_s {freq_hz: 0, freq_offset: 0, if_chain: 0, status: 0, count_us: 0, rf_chain: 0, modem_id: 0, modulation: 0, bandwidth: 0, datarate: 0, coderate: 0, rssic: 0, rssis: 0, snr: 0, snr_min: 0, snr_max: 0, crc: 0, size: 0, payload: [
              0 <repeats 256 times>], ftime_received: false, ftime: 0}]
#6  0x00007ffff75c3e45 in chirpstack_concentratord_sx1302::handler::uplink::handle_loop (gateway_id=..., stop_receive=..., disable_crc_filter=false, time_fallback=false) at chirpstack-concentratord-sx1302/src/handler/uplink.rs:25
[... 97 lines cut ...]

After fix:

```
/* ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ */

int compare_pkt_tmst(const void *a, const void *b)
{
    int p_count, q_count;

    p_count = ((struct lgw_pkt_rx_s *)a)->count_us;
    q_count = ((struct lgw_pkt_rx_s *)b)->count_us;

    return (p_count - q_count);
}

/* ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ */

```

&nbsp;
&nbsp;
&nbsp;

**Let's do IoT better**

LouneCode, 2023-10-04:
`Another hint` <br /> <br /> Also [sx1302_hal/tree/master/packet_forwarder/src
/jitqueue.c](
https://github.com/brocaar/sx1302_hal/blob/master/packet_forwarder/src/jitqueue.c) has a **qsort`_r`()** functionality, similar to  loragw_hal.c.  If there was a special reason to optimize the old sorting method with **qsort()** .  .  .

brocaar, 2023-10-05:
Thank you so much @LouneCode for debugging this issue in so much detail! The reason why this was changed is because `qsort_r` is not available in musl. I took these changes from an other patch, but it seems I took only part of the changes.

I will fix this with the proposed changes. If you would like, I'm also open for a pull-request.

LouneCode, 2023-10-05:
Good that things are progressing....;) <br /> Please, open the pull-request. <br /> FYI now after that fix above, the chirpstack-consentratord-sx1302 process rock like a charm. No more segmentation fault errors.  <br />
You have done a good job. Thanks.

&nbsp;
&nbsp;
&nbsp;

**Let's do IoT better**

brocaar, 2023-10-05:
@LouneCode this commit contains the proposed changes: https://github.com/brocaar/sx1302_hal/commit/c3d99009556fdfe273c3a53306082ef181333c7a. Could you do a final check to make sure this is what you were proposing?

closed 2023-10-05 by brocaar

LouneCode, 2023-10-05:
Thanks for the quick response. Sure, I check it out... <br />
Ok, I have walked thru commits on both repositories. Changes are correct.

I also compiled the latest version of the code of this repository (make build) and did a test run on the chirpstack-concentratord-sx1302 process: `Everything seems fine`.
&nbsp;
&nbsp;
&nbsp;
**Let's do IoT better**

brocaar, 2023-10-05:
Thanks again for your help @LouneCode :+1: