chirpstack/chirpstack-concentratord #114: Crash when receiving configuration command?

cyclops1982, 2023-11-04:
- [X] The issue is present in the latest release.
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## What happened?
chirpstack-concentratord-sx1302 crashes stops/crashes when it receives a 'Configuration' command.

## What did you expect?
That it keeps running :)

## Steps to reproduce this issue
I've installed the chirpstack-concentratord-sx1302 and the mqtt forwarder. I've linked these two together (config's below).
My docker deployed chirpstack is on another machine, so the mqtt forwarder connects to that.
 
It seems that on update from chirpstack, a configuration is passed along to the concentrator and then it crashes.

## Could you share your log output?
[mqtt.log](https://github.com/chirpstack/chirpstack-concentratord/files/13257323/mqtt.log)
[crash.log](https://github.com/chirpstack/chirpstack-concentratord/files/13257324/crash.log)

## Your Environment

forwarder.toml:
```
# Logging settings.
[logging]

  # Log level.
  #
  # Valid options are:
  #   * TRACE
  #   * DEBUG
  #   * INFO
  #   * WARN
  #   * ERROR
  #   * OFF
  level="DEBUG"

  # Log to syslog.
  #
  # If set to true, log messages are being written to syslog instead of stdout.
  log_to_syslog=false

# MQTT settings.
[mqtt]

  # Topic prefix.
  #
  # ChirpStack MQTT Forwarder publishes to the following topics:
  #
  #  * [Prefix/]gateway/[Gateway ID]/event/[Event]
  #  * [Prefix/]gateway/[Gateway ID]/state/[State]
  #
  # And subscribes to the following topic:
  #
  #  * [Prefix/]gateway/[Gateway ID]/command/[Command]
  #
  # The topic prefix can be used to define the region of the gateway.
  # Note, there is no need to add a trailing '/' to the prefix. The trailing
  # '/' is automatically added to the prefix if it is configured.
  topic_prefix="eu868"

  # Use JSON encoding instead of Protobuf (binary).
  #
  # Note, only use this for debugging purposes.
  json=false

  # MQTT server (e.g. scheme://host:port where scheme is tcp, ssl or ws)
  server="tcp://jubilee.lnd.prof-x.net:1883"

  # Connect with the given username (optional)
[... 247 lines cut ...]
    latitude=0.0
    longitude=0.0
    altitude=0

```
### Versions
```
cyclops@octoprint:~/chirpstack-concentrator $ ./chirpstack-mqtt-forwarder  --version
chirpstack-mqtt-forwarder 4.1.2

cyclops@octoprint:~/chirpstack-concentrator $ ./chirpstack-concentratord-sx1302 --version
chirpstack-concentratord-sx1302 4.3.3
```

Chirpstack:
```
[17:32:36] cyclops@jubilee:~$ docker ps -a | grep chirp
9c3fa9351dee   chirpstack/chirpstack-rest-api:4.1.1          "/usr/bin/chirpstack…"   26 minutes ago   Up 26 minutes                                                                                                                                chirpstack_chirpstack-rest-api_1
53df63f34941   chirpstack/chirpstack-gateway-bridge:4.0.3    "/usr/bin/chirpstack…"   26 minutes ago   Up 26 minutes         0.0.0.0:1700->1700/udp                                                                                                 chirpstack_chirpstack-gateway-bridge-eu868_1
e227798d73fa   chirpstack/chirpstack:4.1.1                   "/usr/bin/chirpstack…"   26 minutes ago   Up 26 minutes         0.0.0.0:8088->8080/tcp                                                                                                 chirpstack_chirpstack_1
3524c80ae870   postgres:14-alpine                            "docker-entrypoint.s…"   26 minutes ago   Up 26 minutes         5432/tcp                                                                                                               chirpstack_postgres_1
89f774a33d19   eclipse-mosquitto:2                           "/docker-entrypoint.…"   26 minutes ago   Up 26 minutes         0.0.0.0:1883->1883/tcp                                                                                                 chirpstack_mosquitto_1
548e57d1c80e   redis:7-alpine                                "docker-entrypoint.s…"   26 minutes ago   Up 26 minutes         6379/tcp                                                                                                               chirpstack_redis_1

```

brocaar, 2023-11-10:
The reconfiguration might not be the issue:
> thread '<unnamed>' panicked at 'read from gps error: Error { kind: UnexpectedEof, message: "failed to fill whole buffer" }', chirpstack-concentratord-sx1302/src/handler/gps.rs:75:14

Could you please try without the GPS / GNSS enabled?

cyclops1982, 2023-11-11:
Without GNSS it has been working fine.

I'm wondering how RAK5146 should do GNSS. I have the module that has it built in and the antenna is connected.

brocaar, 2024-04-05:
The module works fine for me. Assuming you are using a Raspberry Pi, have you tested with the ChirpStack Gateway OS? Please note that you need to make some Raspberry Pi config changes to make the GNSS work (by default the Bluetooth module is mapped to the UART pins, which collides with the GNSS).

closed 2024-05-06 by brocaar