chirpstack/chirpstack-concentratord #123: Concentratord overrides static/dynamic metadata defined in mqtt-forwarder configuration file

nicolas-juteau, 2023-12-06:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## What happened?
MQTT forwarder metadata defined gets overriden by concentratord (either static or dynamic metadata). NOTE: commands are not affected by this issue (one can execute custom commands as defined in mqtt-forwarder, whether used backend is concentratord or semtech_udp

## What did you expect?
I would expect that the static/dynamic metadata defined in mqtt-forwarder appears along with concentratord self-defined metadata in Chirpstack Gateways -> Configuration -> Metadata page. 

## Steps to reproduce this issue

Steps:

1. Define some metadata (either static and/or dynamic) in mqtt-forwarder and configure it so it uses concentratord backend
2. Run mqtt-forwarder along with concentratord on the gateway
3. In Chirpstack, access the gateway configuration / metadata page. The custom defined metadata does not appear, only concentratord-related metadata
4. Now modify mqtt-forwarder so it uses semtech_udp backend.
5. Restart mqtt-forwarder & semtech packet forwarder
6. In Chirpstack, access the gateway configuration / metadata page. The custom defined metadata will appear as expected.

## Could you share your log output?

No log file involved.

## Your Environment

Environment all uses latest version (v4) of chirpstack, chirpstack-concentratord and chirpstack-mqtt-forwarder

nicolas-juteau, 2023-12-06:
FYI: gateway used is Multitech Conduit (MTCDT-0.1).

brocaar, 2023-12-07:
Fixed in the above commit.

closed 2023-12-07 by brocaar