chirpstack/chirpstack-concentratord #127: MTCAP3: wrong gateway_id reported by concentratord to mqtt-forwarder

nicolas-juteau, 2023-12-13:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## What happened?
Using MTCAP3 gateway and the latest concentratord/mqtt-forwarder chirpstack pre-compiled packages, the gateway_id reported by the concentratord (or read by the mqtt-forwarder) does not match the configured gateway_id inside concentratord.toml configuration file:

> gateway_id="0000000000000000"

After starting concentratord/mqtt-forward processes, the following can be seen in the logs:

> 2023-12-13T13:41:21.293871+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Starting the concentrator
> 2023-12-13T13:41:23.807940+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Gateway ID retrieved, gateway_id: "0016c001f1388eb8"
> 2023-12-13T13:44:49.899489+00:00 mtcap3 chirpstack-mqtt-forwarder[1793]: Reading gateway id
> 2023-12-13T13:44:49.913138+00:00 mtcap3 chirpstack-mqtt-forwarder[1793]: Received gateway id, gateway_id: 0016c001f1388eb8
> 2023-12-13T13:44:49.914460+00:00 mtcap3 chirpstack-mqtt-forwarder[1793]: Received Gateway ID from backend, gateway_id: 0016c001f1388eb8

This does not occur on a MTCDT gateway.

## What did you expect?
I would expect the reported gateway id by concentratord/mqtt-forwarder to matches the gateway id as defined in the concentratord.toml configuration file on the MTCAP3 gateway.

## Steps to reproduce this issue

Steps:

1. On the MTCAP3, setup a mqtt-forwarder + concentratord.
2. Set the gateway_id parameter to whatever value in concentratord.toml configuration file
3. Restart the mqtt-forwarder / concentratord processes.
4. Inspect the logs, gateway id reported by concentratord / mqtt-forwarder will not match the one defined in concentratord.toml configuration file

## Could you share your log output?

2023-12-13T13:41:54.072747+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Stopping the concentrator
2023-12-13T13:41:54.079098+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Updating concentrator configuration
2023-12-13T13:41:54.080469+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Starting Concentratord SX1302 (version: 4.3.4, docs: https://www.chirpstack.io/docs/chirpstack-concentratord/)
2023-12-13T13:41:54.081765+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Executing reset command, command: mts-io-sysfs, args: ["store", "lora/reset", "0"]
2023-12-13T13:41:54.219920+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Executing reset command, command: mts-io-sysfs, args: ["store", "lora/reset", "1"]
2023-12-13T13:41:54.361003+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Setting i2c device path, path: /dev/i2c-1
2023-12-13T13:41:54.361362+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Setting board configuration, lorawan_public: true, clock_source: 0, com_type: Spi, com_path: /dev/spidev1.0
2023-12-13T13:41:54.361648+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Setting up fine timestamp, enable: false
2023-12-13T13:41:54.361892+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Setting up concentrator channels
2023-12-13T13:41:54.362205+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring radio, radio: 0, enabled: true, center_freq: 902700000, type: SX1250
2023-12-13T13:41:54.362472+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring radio, radio: 1, enabled: true, center_freq: 903700000, type: SX1250
2023-12-13T13:41:54.362691+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Setting up concentrator channels
2023-12-13T13:41:54.362993+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring multi-SF LoRa channel, channel: 0, enabled: true, freq: 902300000, rf_chain: 0, if_freq: -400000
2023-12-13T13:41:54.363272+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring multi-SF LoRa channel, channel: 1, enabled: true, freq: 902500000, rf_chain: 0, if_freq: -200000
2023-12-13T13:41:54.363541+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring multi-SF LoRa channel, channel: 2, enabled: true, freq: 902700000, rf_chain: 0, if_freq: 0
2023-12-13T13:41:54.363795+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring multi-SF LoRa channel, channel: 3, enabled: true, freq: 902900000, rf_chain: 0, if_freq: 200000
2023-12-13T13:41:54.368515+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring multi-SF LoRa channel, channel: 4, enabled: true, freq: 903100000, rf_chain: 0, if_freq: 400000
2023-12-13T13:41:54.369768+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring multi-SF LoRa channel, channel: 5, enabled: true, freq: 903300000, rf_chain: 1, if_freq: -400000
2023-12-13T13:41:54.371004+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring multi-SF LoRa channel, channel: 6, enabled: true, freq: 903500000, rf_chain: 1, if_freq: -200000
2023-12-13T13:41:54.372219+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring multi-SF LoRa channel, channel: 7, enabled: true, freq: 903700000, rf_chain: 1, if_freq: 0
2023-12-13T13:41:54.373417+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring Std LoRa channel, enabled: true, freq: 903000000, rf_chain: 0, if_freq: 300000
2023-12-13T13:41:54.374594+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Configuring FSK channel, enabled: false, freq: 0, rf_chain: 0, if_freq: 0
2023-12-13T13:41:54.376791+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Starting the concentrator
2023-12-13T13:41:56.882896+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Gateway ID retrieved, gateway_id: "0016c001f1388eb8"
2023-12-13T13:41:56.884252+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Initializing JIT queue, capacity: 32
2023-12-13T13:41:56.885795+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Creating socket for publishing events, bind: ipc:///tmp/concentratord_event
2023-12-13T13:41:56.892755+00:00 mtcap3 chirpstack-concentratord-sx1302[1521]: Creating socket for receiving commands, bind: ipc:///tmp/concentratord_command
2023-12-13T13:44:29.661536+00:00 mtcap3 chirpstack-mqtt-forwarder[1744]: Starting ChirpStack MQTT Forwarder (version: 4.1.3, docs: https://www.chirpstack.io/)
2023-12-13T13:44:29.662329+00:00 mtcap3 chirpstack-mqtt-forwarder[1744]: Setting up ChirpStack Concentratord backend
2023-12-13T13:44:29.663187+00:00 mtcap3 chirpstack-mqtt-forwarder[1744]: Setting up ChirpStack Concentratord backend
2023-12-13T13:44:29.665181+00:00 mtcap3 chirpstack-mqtt-forwarder[1744]: Connecting to Concentratord event API, event_url: ipc:///tmp/concentratord_event
2023-12-13T13:44:29.667677+00:00 mtcap3 chirpstack-mqtt-forwarder[1744]: Connecting to Concentratord command API, command_url: ipc:///tmp/concentratord_command
2023-12-13T13:44:29.668930+00:00 mtcap3 chirpstack-mqtt-forwarder[1744]: Reading gateway id
2023-12-13T13:44:29.676716+00:00 mtcap3 chirpstack-mqtt-forwarder[1744]: Received gateway id, gateway_id: 0016c001f1388eb8
2023-12-13T13:44:29.677987+00:00 mtcap3 chirpstack-mqtt-forwarder[1744]: Received Gateway ID from backend, gateway_id: 0016c001f1388eb8

## Your Environment

| Component           | Version |
| --------------------| ------- |
| MQTT Forwarder  |  4.1.3       |
| Concentratord       |  4.3.4       |

brocaar, 2023-12-14:
The SX1302 has the gateway-id baked in the chip, there is no `gateway_id` configuration option:
https://www.chirpstack.io/docs/chirpstack-concentratord/configuration.html#configuration-example-1

> Unlike the -sx1301 binary, there is no option for configuring the gateway_id. The unique gateway ID is embedded in the SX1302 and read by the Concentratord daemon.

nicolas-juteau, 2023-12-14:
Oh, my bad. Good to know. Thank you!