chirpstack/chirpstack-concentratord #152: For SPI's dev device address and resetting GPIO errors

TelDragon, 2024-04-24:
Dear author, hello.
I am using Raspberry Pi 4 product. I want to implement the lorawan function.
Currently, it is running using lora_pkt_fwd
I have noticed your subroutine, but I have identified a small issue
It defaults to using the spi dev address of/dev/spidev0.0, and the reset address is 7
But my/dev/spidev0.0 and gpio 7 have already been occupied.
How can it be customized and changed in the toml configuration file?

error log info

```

./chirpstack-concentratord-sx1302 -c chirpstack-concentratord-sx1302.toml 
2024-04-24T17:09:28.682Z INFO  [libconcentratord::reset] Configuring reset pin, dev: /dev/gpiochip0, pin: 17
2024-04-24T17:09:28.682Z INFO  [chirpstack_concentratord_sx1302::cmd::root] Starting Concentratord SX1302 (version: 4.3.5, docs: https://www.chirpstack.io/docs/chirpstack-concentratord/)
2024-04-24T17:09:28.683Z INFO  [libconcentratord::reset] Triggering sx1302 reset
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Setting i2c device path, path: /dev/i2c-1
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Setting board configuration, lorawan_public: true, clock_source: 0, com_type: Spi, com_path: /dev/spidev0.0
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Setting up fine timestamp, enable: false
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Setting up concentrator channels
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring radio, radio: 0, enabled: true, center_freq: 867500000, type: SX1250
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring radio, radio: 1, enabled: true, center_freq: 868500000, type: SX1250
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Setting up concentrator channels
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring multi-SF LoRa channel, channel: 0, enabled: true, freq: 868100000, rf_chain: 1, if_freq: -400000
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring multi-SF LoRa channel, channel: 1, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring multi-SF LoRa channel, channel: 2, enabled: true, freq: 868500000, rf_chain: 1, if_freq: 0
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring multi-SF LoRa channel, channel: 3, enabled: true, freq: 867100000, rf_chain: 0, if_freq: -400000
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring multi-SF LoRa channel, channel: 4, enabled: true, freq: 867300000, rf_chain: 0, if_freq: -200000
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring multi-SF LoRa channel, channel: 5, enabled: true, freq: 867500000, rf_chain: 0, if_freq: 0
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring multi-SF LoRa channel, channel: 6, enabled: true, freq: 867700000, rf_chain: 0, if_freq: 200000
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring multi-SF LoRa channel, channel: 7, enabled: true, freq: 867900000, rf_chain: 0, if_freq: 400000
2024-04-24T17:09:28.883Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring Std LoRa channel, enabled: true, freq: 868300000, rf_chain: 1, if_freq: -200000
2024-04-24T17:09:28.884Z INFO  [chirpstack_concentratord_sx1302::concentrator] Configuring FSK channel, enabled: true, freq: 868800000, rf_chain: 1, if_freq: 300000
2024-04-24T17:09:28.884Z INFO  [chirpstack_concentratord_sx1302::concentrator] Starting the concentrator
Opening SPI communication interface
Note: chip version is 0x00 (v0.0)
ERROR: Failed to set SX1250_0 in STANDBY_RC mode
ERROR: failed to setup radio 0
thread 'main' panicked at chirpstack-concentratord-sx1302/src/main.rs:117:80:
called `Result::unwrap()` on an `Err` value: lgw_start failed
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

```

brocaar, 2024-04-30:
What shield are you using and how is it connected to your Raspberry Pi?

TelDragon, 2024-05-07:
> What shield are you using and how is it connected to your Raspberry Pi?

raspberry pi (cm4)  - boot/config.txt
```
dtoverlay=spi0-1cs
dtoverlay=spi1-1cs
```

dev  list
```
ls -ll /dev/spidev*
crw-rw---- 1 root spi 153, 0 May  6 16:30 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 May  6 16:30 /dev/spidev1.0
```

spidev0.0 For my 14 inch LCD screen ：

VCC ==  5V == 5V 
GND == GND == GND
DIN == MOSI == 19
CLK == SCLK == 23
CS == CE0 == 24
DC == 25 == 22
RST == 27 == 13
BL == 18 == 12

spidev1.0  SX1302 Inside

brocaar, 2024-05-07:
But which brand / model concentrator shield do you have?

TelDragon, 2024-05-07:
> But which brand / model concentrator shield do you have?
> […](#)

For the expansion board connecting Raspberry Pi CM4, we create our own expansion board to supplement our products. Not using expansion boards from mature and well-known brands in the market.

brocaar, 2024-05-07:
> we create our own expansion board to supplement our products

In that case, you might want to make some modifications as the Concentratord keeps track of each Vendor / model profile. Please note that different gateway models might have different calibration values. These calibration tables make sure that when the gateway is requested to transmit with tx power X, that tx power X is feed into the antenna. Example:

https://github.com/chirpstack/chirpstack-concentratord/blob/master/chirpstack-concentratord-sx1302/src/config/vendor/rak/rak5146.rs#L45

If you decide to use an existing model, then you can override the reset pins using the following options:

https://github.com/chirpstack/chirpstack-concentratord/blob/master/chirpstack-concentratord-sx1302/src/config/mod.rs#L99

E.g. under the `[gateway]` section in the TOML config, you can add:

```toml
[gateway]
sx1302_reset_chip="/dev/gpiochip0"
sx1302_reset_pin=17
sx1302_power_en_chip="/dev/gpiochip0"
sx1302_power_en_pin=18
```

closed 2024-05-07 by brocaar