chirpstack/chirpstack-concentratord #167: Process panicks upon start (Multitech Conduit, mPower 6.3.0)

nicolas-juteau, 2024-06-25:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## What happened?
Attempted to start chirpstack-concentratord 4.4.1 on a Multitech Conduit (LoRa Card MTAC-LORA-H-915) with mPower 6.3.0 version. The process crashes and does not start.

## What did you expect?
I did expect the concentratord to start as it should.

## Steps to reproduce this issue

Steps:

1. Install mPower 6.3.0 on a Multitech Conduit gateway (model used is MTCDT-L4N1)
2. Install chirpstack-concentratord, following the instructions as indicated in documentation (https://www.chirpstack.io/docs/chirpstack-concentratord/installation/multitech.html)
3. Attempt to start chirpstack-concentratord and notice it does not fork in the background (process exits)
4. See the logs / command output

## Could you share your log output?

```shell
root@mtcdt:/var/config/chirpstack-concentratord-sx1301/ap2# /opt/chirpstack-concentratord-sx1301/chirpstack-concentratord-sx1301 -c concentratord.toml -c channels.toml -c region.toml
thread 'main' panicked at chirpstack-concentratord-sx1301/src/main.rs:111:80:
called `Result::unwrap()` on an `Err` value: lgw_start failed
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
root@mtcdt:/var/config/chirpstack-concentratord-sx1301/ap2# RUST_BACKTRACE=1 /opt/chirpstack-concentratord-sx1301/chirpstack-concentratord-sx1301 -c concentratord.toml -c channels.toml -c region.toml
thread 'main' panicked at chirpstack-concentratord-sx1301/src/main.rs:111:80:
called `Result::unwrap()` on an `Err` value: lgw_start failed

Stack backtrace:
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
stack backtrace:
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
root@mtcdt:/var/config/chirpstack-concentratord-sx1301/ap2# RUST_BACKTRACE=full /opt/chirpstack-concentratord-sx1301/chirpstack-concentratord-sx1301 -c concentratord.toml -c channels.toml -c region.toml
thread 'main' panicked at chirpstack-concentratord-sx1301/src/main.rs:111:80:
called `Result::unwrap()` on an `Err` value: lgw_start failed

Stack backtrace:
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
stack backtrace:
   0:   0x29200c - <unknown>
   1:   0x2dced8 - <unknown>
   2:   0x28ea40 - <unknown>
   3:   0x291e40 - <unknown>
   4:   0x29366c - <unknown>
   5:   0x2932bc - <unknown>
   6:   0x293ba0 - <unknown>
   7:   0x2939fc - <unknown>
   8:   0x292554 - <unknown>
   9:   0x2937c4 - <unknown>
  10:    0x1e8b8 - <unknown>
  11:    0x1ed04 - <unknown>
  12:    0x554f0 - <unknown>
  13:    0x73e18 - <unknown>
  14:    0x81770 - <unknown>
  15:   0x2887b8 - <unknown>
  16:    0x561d4 - <unknown>
root@mtcdt:/var/config/chirpstack-concentratord-sx1301/ap2#

2024-06-24T16:49:22.323623+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Starting Concentratord SX1301 (version: 4.4.1, docs: https://www.chirpstack.io/docs/chirpstack-concentratord/)
2024-06-24T16:49:22.329014+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Setting spi device path, spidev_path: /dev/spidev32765.2
2024-06-24T16:49:22.334618+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Setting board configuration, lorawan_public: true, clock_source: 0
2024-06-24T16:49:22.341344+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Setting up concentrator radios
2024-06-24T16:49:22.345681+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring radio, radio: 0, enabled: true, center_freq: 902700000, type: SX1257
2024-06-24T16:49:22.347451+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring radio, radio: 1, enabled: true, center_freq: 903700000, type: SX1257
2024-06-24T16:49:22.349203+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Setting up concentrator channels
2024-06-24T16:49:22.351008+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring multi-SF LoRa channel, channel: 0, enabled: true, freq: 902300000, rf_chain: 0, if_freq: -400000
2024-06-24T16:49:22.353912+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring multi-SF LoRa channel, channel: 1, enabled: true, freq: 902500000, rf_chain: 0, if_freq: -200000
2024-06-24T16:49:22.355835+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring multi-SF LoRa channel, channel: 2, enabled: true, freq: 902700000, rf_chain: 0, if_freq: 0
2024-06-24T16:49:22.357630+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring multi-SF LoRa channel, channel: 3, enabled: true, freq: 902900000, rf_chain: 0, if_freq: 200000
2024-06-24T16:49:22.359434+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring multi-SF LoRa channel, channel: 4, enabled: true, freq: 903100000, rf_chain: 0, if_freq: 400000
2024-06-24T16:49:22.361229+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring multi-SF LoRa channel, channel: 5, enabled: true, freq: 903300000, rf_chain: 1, if_freq: -400000
2024-06-24T16:49:22.363027+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring multi-SF LoRa channel, channel: 6, enabled: true, freq: 903500000, rf_chain: 1, if_freq: -200000
2024-06-24T16:49:22.365055+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring multi-SF LoRa channel, channel: 7, enabled: true, freq: 903700000, rf_chain: 1, if_freq: 0
2024-06-24T16:49:22.366720+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring Std LoRa channel, enabled: true, freq: 903000000, rf_chain: 0, if_freq: 300000
2024-06-24T16:49:22.368524+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Configuring FSK channel, enabled: false, freq: 0, rf_chain: 0, if_freq: 0
2024-06-24T16:49:22.370096+00:00 mtcdt chirpstack-concentratord-sx1301[3828]: Starting the concentrator
```

_spidev_path: /dev/spidev32765.2_ looks odd to me.

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Concentratord       |  4.4.1        |

nicolas-juteau, 2024-06-25:
Also, I am able to run the embedded lora-network-server + packet forwarder provided by the mPower, so it is apparently not a LoRa card issue.

nicolas-juteau, 2024-06-26:
Looks like to be confirmed in the mLinux changelog:
[image]

brocaar, 2024-08-30:
Sorry for the delay, I haven't had the time yet to validate this. What I will do is update my Conduit to the latest firmware version and then update the SPI dev. paths where needed. Note that in case needed, there is always an option to override this path in the `concentratord.toml` file (e.g. https://github.com/chirpstack/chirpstack-concentratord/blob/master/chirpstack-concentratord-sx1302/src/config/mod.rs#L109).

commit 170d0db3 (2024-08-30): sx1301: Update multitech_mtac_lora_h_915 SPI path.

The SPI path has been updated since the mPower 6.x firmware, but this
change was not applied to this config (it was fixed for _868).

Closes: #167.
closed 2024-08-30 by brocaar