chirpstack/chirpstack-concentratord #184: GNSS override is ignored for waveshare_sx1302_lorawan_gateway_hat

errolt, 2024-10-13:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## What happened?
Setting  model_flags=[    "GNSS"  ] is ignored when model="waveshare_sx1302_lorawan_gateway_hat"

## What did you expect?
I expected the GPS to be used.
If I set model="rak_2287", with the waveshare radio still connected, then the GPS is used, even though the GPS on the module isn't UBLOX

## Steps to reproduce this issue
Steps:
Add GNSS for model_flags, set model to waveshare_sx1302_lorawan_gateway_hat and set  log_level="DEBUG"

## Could you share your log output?

With model="waveshare_sx1302_lorawan_gateway_hat"
```shell
Opening SPI communication interface
Note: chip version is 0x12 (v1.2)
INFO: Configuring SX1250_0 in single input mode
INFO: using legacy timestamp
INFO: LoRa Service modem: configuring preamble size to 8 symbols
ARB: dual demodulation disabled for all SF
2024-10-13T10:46:40.964Z DEBUG [chirpstack_concentratord_sx1302::concentrator] Getting gateway EUI
2024-10-13T10:46:40.965Z INFO  [chirpstack_concentratord_sx1302::cmd::root] Gateway ID retrieved, gateway_id: "0016c001f15d6607"
2024-10-13T10:46:40.965Z INFO  [libconcentratord::jitqueue] Initializing JIT queue, capacity: 32
2024-10-13T10:46:40.965Z INFO  [libconcentratord::events] Creating socket for publishing events, bind: ipc:///tmp/concentratord_event
2024-10-13T10:46:40.966Z INFO  [libconcentratord::commands] Creating socket for receiving commands, bind: ipc:///tmp/concentratord_command
2024-10-13T10:46:40.966Z DEBUG [chirpstack_concentratord_sx1302::handler::uplink] Starting uplink handle loop
2024-10-13T10:46:40.967Z DEBUG [chirpstack_concentratord_sx1302::handler::jit] Starting JIT queue loop
2024-10-13T10:46:40.967Z DEBUG [chirpstack_concentratord_sx1302::handler::command] Starting command handler loop
2024-10-13T10:46:40.967Z DEBUG [chirpstack_concentratord_sx1302::handler::stats] Starting stats loop, stats_interval: 30s
```

With model="rak_2287"
```shell
Opening SPI communication interface
Note: chip version is 0x12 (v1.2)
INFO: Configuring SX1250_0 in single input mode
INFO: using legacy timestamp
INFO: LoRa Service modem: configuring preamble size to 8 symbols
ARB: dual demodulation disabled for all SF
2024-10-13T10:37:25.920Z DEBUG [chirpstack_concentratord_sx1302::concentrator] Getting gateway EUI
2024-10-13T10:37:25.922Z INFO  [chirpstack_concentratord_sx1302::cmd::root] Gateway ID retrieved, gateway_id: "0016c001f15d6607"
2024-10-13T10:37:25.923Z INFO  [libconcentratord::jitqueue] Initializing JIT queue, capacity: 32
2024-10-13T10:37:25.923Z INFO  [libconcentratord::events] Creating socket for publishing events, bind: ipc:///tmp/concentratord_event
2024-10-13T10:37:25.924Z INFO  [libconcentratord::commands] Creating socket for receiving commands, bind: ipc:///tmp/concentratord_command
2024-10-13T10:37:25.925Z DEBUG [chirpstack_concentratord_sx1302::handler::uplink] Starting uplink handle loop
2024-10-13T10:37:25.925Z DEBUG [chirpstack_concentratord_sx1302::handler::jit] Starting JIT queue loop
2024-10-13T10:37:25.926Z DEBUG [chirpstack_concentratord_sx1302::handler::command] Starting command handler loop
2024-10-13T10:37:25.926Z DEBUG [chirpstack_concentratord_sx1302::handler::stats] Starting stats loop, stats_interval: 30s
2024-10-13T10:37:25.927Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Starting GPS loop
2024-10-13T10:37:25.927Z INFO  [chirpstack_concentratord_sx1302::handler::gps] Enabling GPS device, tty_path: /dev/ttyS2
2024-10-13T10:37:25.929Z INFO  [chirpstack_concentratord_sx1302::handler::gps] Starting GPS validation loop
2024-10-13T10:37:26.867Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-13T10:37:26.929Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1728815846.929692814s
2024-10-13T10:37:27.000Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
2024-10-13T10:37:27.002Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] GPS sync error, error: lgw_gps_sync failed
2024-10-13T10:37:27.867Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-13T10:37:27.930Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1728815847.930099601s
2024-10-13T10:37:28.000Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
2024-10-13T10:37:28.001Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] GPS sync error, error: lgw_gps_sync failed
2024-10-13T10:37:28.861Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-13T10:37:28.930Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1728815848.930370765s
2024-10-13T10:37:28.994Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
2024-10-13T10:37:29.866Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-13T10:37:29.999Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
2024-10-13T10:37:30.863Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-13T10:37:30.996Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
2024-10-13T10:37:31.859Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-13T10:37:31.991Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
2024-10-13T10:37:32.861Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-13T10:37:32.990Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v?.?.?  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       | v4.4.3 |

errolt, 2024-10-13:
Side note. This is running on a Radxa Rock 5B board. Hence the strange GPS port, i2c port and GPIO allocations.

If this is an issue proving that the GPS on the waveshare hat works fine, even though it isn't Ublox, then I'll be happy to set up a PI4 to do the same tests.

brocaar, 2024-10-14:
The Waveshare shield does not have an U-blox GNSS module. Please note that the Semtech HAL only supports the  U-blox GNSS modules because it relies on a proprietary binary message for accurate time syncing.

errolt, 2024-10-14:
@brocaar Did you read my complete report? The GPS module on the waveshare hat **WORKS** if I tell concentratord that I have a rak_2287. Does changing the model in the config magically change the GPS module to a UBLOX module?

As I noted in another issue a few years ago, the waveshare GNSS module has been modified to emit UBLOX time messages.

So in the end I just have to run my concentrator misconfigured, set up as a rak_2287, with incorrect power calibration or whatever that causes, just to get GPS to work?

brocaar, 2024-10-28:
I have tested with my Waveshare module to reproduce what you are saying, but it does not work for me. The Waveshare module only reports NMEA RMC messages. If it supports the NAV-TIMEGPS (which is a proprietary u-blox msg), it should have printed this in the logs every second and it does not:

https://github.com/chirpstack/chirpstack-concentratord/blob/master/chirpstack-concentratord-sx1302/src/handler/gps.rs#L98

This is what the Concentratord outputs:

```
2024-10-28T12:27:04.512Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Starting GPS loop
2024-10-28T12:27:04.512Z INFO  [chirpstack_concentratord_sx1302::handler::gps] Enabling GPS device, tty_path: /dev/ttyAMA0
2024-10-28T12:27:04.512Z INFO  [chirpstack_concentratord_sx1302::handler::gps] Starting GPS validation loop
2024-10-28T12:27:05.206Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:05.206Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] get gps coordinates failed, error: lgw_gps_get failed
2024-10-28T12:27:05.512Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118425.512831649s
2024-10-28T12:27:06.205Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:06.513Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118426.513058794s
2024-10-28T12:27:07.196Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:07.513Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118427.513292139s
2024-10-28T12:27:08.200Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:08.513Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118428.513531554s
2024-10-28T12:27:09.198Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:09.512Z DEBUG [chirpstack_concentratord_sx1302::handler::beacon] Get GPS epoch error, error: gps time reference not available
2024-10-28T12:27:09.513Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118429.513681444s
2024-10-28T12:27:10.197Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:10.513Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118430.513882843s
2024-10-28T12:27:11.205Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:11.514Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118431.514056621s
2024-10-28T12:27:12.205Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:12.514Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118432.514207036s
2024-10-28T12:27:13.197Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:13.514Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118433.514356456s
2024-10-28T12:27:14.196Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:14.514Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118434.514591285s
2024-10-28T12:27:15.200Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:15.512Z DEBUG [chirpstack_concentratord_sx1302::handler::beacon] Get GPS epoch error, error: gps time reference not available
2024-10-28T12:27:15.514Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118435.51482584s
2024-10-28T12:27:16.196Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:16.515Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118436.515028858s
2024-10-28T12:27:17.196Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
2024-10-28T12:27:17.515Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1730118437.515224301s
```

Config:

```toml
[concentratord]
log_level="DEBUG"
log_to_syslog=false
stats_interval="30s"
disable_crc_filter=false

[concentratord.api]
[gateway]
lorawan_public=true
model="waveshare_sx1302_lorawan_gateway_hat"
region="EU868"
model_flags=["GNSS",]
antenna_gain=2
```

Tested with a custom Concentratord that enables GPS for the Waveshare module.

I can not find a single source that confirms that the GNSS module should support this:
https://www.google.com/search?q=quelctel+l76+%22NAV-TIMEGPS%22

closed 2024-10-28 by brocaar

errolt, 2024-10-28:
Strange. As that is exactly what I see:

```
Oct 28 15:01:30 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:30.772Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:30 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:30.907Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:31 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:31.762Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:31 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:31.900Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:32 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:32.772Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:32 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:32.907Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:33 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:33.766Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:33 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:33.902Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:34 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:34.766Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:34 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:34.901Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:35 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:35.776Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:35 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:35.911Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:36 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:36.776Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:36 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:36.912Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:37 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:37.776Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:37 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:37.912Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:38 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:38.779Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:38 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:38.914Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:39 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:39.767Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:39 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:39.903Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:40 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:40.769Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:40 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:40.905Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:41 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:41.774Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:41 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:41.909Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:42 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:42.775Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:42 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:42.910Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:43 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:43.769Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:43 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:43.905Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:44 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:44.770Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:44 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:44.906Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:45 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:45.773Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:45 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:45.908Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
[... cut ...]
Oct 28 15:01:52 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:52.926Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
Oct 28 15:01:53 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:53.779Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing NMEA RMC
Oct 28 15:01:53 rock-5b chirpstack-concentratord-sx1302[949]: 2024-10-28T13:01:53.915Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Processing u-blox NAV-TIMEGPS
```

Sample output for my gps:
[gps_sample.txt](https://github.com/user-attachments/files/17542840/gps_sample.txt)

errolt, 2024-10-28:
Note that I got that output with:
```
 model="rak_2287"
 gnss_dev_path="/dev/ttyAMA0"
 com_dev_path="/dev/spidev0.0"
 sx1302_reset_pin=23
 sx1302_power_en_pin=18`

```
And my config:
```
# Concentratord configuration.
[concentratord]
  # Log level.
  #
  # Valid options are:
  #   * TRACE
  #   * DEBUG
  #   * INFO
  #   * WARN
  #   * ERROR
  #   * OFF
  log_level="DEBUG"

  # Log to syslog.
  #
  # When set to true, log messages are being written to syslog instead of stdout.
  log_to_syslog=false

  # Statistics interval.
  stats_interval="30s"

  # Disable CRC status filter.
  #
  # By default, the Concentratord will ignore received frames which do not have
  # a valid CRC. This option makes it possible to disable this filter such that
  # received frames without (valid) CRC can be analyzed.
  disable_crc_filter=false

  # Configuration for the (ZeroMQ based) API.
  [concentratord.api]
    # Event PUB socket bind.
    event_bind="ipc:///tmp/concentratord_event"

    # Command REP socket bind.
    command_bind="ipc:///tmp/concentratord_command"

# LoRa gateway configuration.
[gateway]

  # Antenna gain (dBi).
  antenna_gain=0

  # Public LoRaWAN network.
  lorawan_public=true

  # Region.
  #
  # The region of the gateway. Options:
  #  EU868, US915, CN779, EU433, AU915, CN470, AS923, AS923_2, AS923_3, AS923_4,
  #  KR923, IN865, RU864
  #
  # Not not all the gateway models implement all regions.
  region="EU868"

  # Gateway vendor / model.
  #
  # This configures various vendor and model specific settings.
  #model="rak_2287"
  #model="waveshare_sx1302_lorawan_gateway_hat"
  model="rak_2287"

  # Gateway vendor / model flags.
  #
  # Flag can be used to configure additional vendor / model features. The
  # following flags can be used:
  #
  #   Global flags:
  #     GNSS - Enable GNSS / GPS support
  #     USB  - Use USB for concentrator communication (default is SPI)
  model_flags=[
    "GNSS"
  ]

  # Time fallback.
  #
  # In case the gateway does not have a GNSS module or is unable to aquire a
  # GNSS fix, use the system-time for setting the 'time' field on RX.
  time_fallback_enabled=true

  # Device and pin configuration.
  #
  # Only set this configuration if you need to override the default
  # configuration provided by the model configuration and you know
  # the device and pin mapping. In any other case, leave this commented
  # out.
  gnss_dev_path="/dev/ttyAMA0"
  # gnss_dev_path="gpsd://localhost:2947"
  com_dev_path="/dev/spidev0.0"
  #i2c_dev_path="/dev/i2c-7"
  #sx1302_reset_chip="/dev/gpiochip3"
  sx1302_reset_pin=23
  #sx1302_power_en_chip="/dev/gpiochip3"
  sx1302_power_en_pin=18

  # LoRa concentrator configuration.
  [gateway.concentrator]

    # Multi spreading-factor channels (LoRa).
    multi_sf_channels=[
      868100000,
      868300000,
      868500000,
      867100000,
      867300000,
      867500000,
      867700000,
      867900000,
    ]

    # LoRa std channel (single spreading-factor).
    [gateway.concentrator.lora_std]
      frequency=868300000
      bandwidth=250000
      spreading_factor=7

    # FSK channel.
    [gateway.concentrator.fsk]
      frequency=868800000
      bandwidth=125000
      datarate=50000

  # Static gateway location.
  [gateway.location]

    # When set to non-zero values, the static gateway location will be reported
    # when the gateway does not have a GNSS module or when no GNSS location fix
    # is available.
    latitude=0.0
    longitude=0.0
    altitude=0
```

brocaar, 2024-10-28:
Just to be sure, I tried with your config but I can not reproduce the same output with NAV-TIMEGPS using your config.

Maybe your shield is from a different batch, that I don't know. However to support GPS I would like to see a confirmation from Waveshare that this is expected to work. Mine has a Quectel L76 module and I can find zero references that this module is supposed to support the NAV-TIMEGPS binary messages.

See below the picture of my shield + GNSS module:

[image]

[image]

errolt, 2024-10-28:
I have not looked at my older hat, but the new hat on my desk has a Q76K, which might have the ublox addition?
[image]