chirpstack/chirpstack-concentratord #194: Fail to build on Apple M1

jerryyip, 2024-11-13:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

## What happened?

I failed to build the concentrated(armv7) on my apple M1 MacBook.

## What did you expect?

## Steps to reproduce this issue

Steps:

1. Active `nix-shell`
2. Run `make build-armv7-unknown-linux-musleabihf` 

## Could you share your log output?
```shell
$ make build-armv7-unknown-linux-musleabihf
cross build --target armv7-unknown-linux-musleabihf --release
[+] Building 2.4s (3/3) FINISHED                                                                                                                                                                                                                
 => [internal] load build definition from Dockerfile.armv7-unknown-linux-musleabihf                                                                                                                                                        0.0s
 => => transferring dockerfile: 68B                                                                                                                                                                                                        0.0s
 => [internal] load .dockerignore                                                                                                                                                                                                          0.0s
 => => transferring context: 2B                                                                                                                                                                                                            0.0s
 => ERROR [internal] load metadata for ghcr.io/cross-rs/armv7-unknown-linux-musleabihf:main                                                                                                                                                2.4s
------
 > [internal] load metadata for ghcr.io/cross-rs/armv7-unknown-linux-musleabihf:main:
------
failed to solve with frontend dockerfile.v0: failed to create LLB definition: no match for platform in manifest sha256:89b44262cc96476b38c296940ce023d3dee98748cf35fd4b590d76b8d5908236: not found
```

## Your Environment

- Apple M1 Pro(with masOS 14.1.2 )

| Component           | Version |
| --------------------| ------- |
| Application Server  | v?.?.?  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

brocaar, 2024-11-13:
I'm not sure if cross-rs supports other platforms than linux/amd64 at the moment, see:
https://github.com/cross-rs/cross/pkgs/container/armv7-unknown-linux-musleabihf

[image]

jerryyip, 2024-11-14:
Yes, I confirm that `unknown/unknown` doesn't work on Apple M1.

closed 2024-11-14 by jerryyip