chirpstack/chirpstack-concentratord #197: support  build-mipsel-unknown-linux-musl

DanielCao0, 2024-11-20:
Can add a new target ? build-mipsel-unknown-linux-musl. Thank you.

brocaar, 2024-11-21:
Hi @DanielCao0, do you mean just in the `Makefile`, or also distributing `.ipk` packages? I assume your request would be for the RAK target is that correct (like for the `chirpstack-mqtt-forwarder`)?

DanielCao0, 2024-11-22:
> Hi @DanielCao0, do you mean just in the `Makefile`, or also distributing `.ipk` packages? I assume your request would be for the RAK target is that correct (like for the `chirpstack-mqtt-forwarder`)?

Yes, I mean just Makefile add mipsel (RAK)  target.   (like for the chirpstack-mqtt-forwarder and mesh)

brocaar, 2024-11-26:
Hi @DanielCao0 please see the above commit.

commit 8084cbda (2024-11-26): Add build-mipsel-unknown-linux-musl task.

Closes #197.
closed 2024-11-26 by brocaar