chirpstack/chirpstack-concentratord #198: A new Gpsd property – never ending story...

LouneCode, 2024-11-22:
- [x ] The issue is present in the latest release.
- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-concentratord) of this repository and believe that this is not a duplicate.

</BR>

## Concentratord process seems to hang in following situation:
 [Gateway] section in the configuration file:
- model_flags=["USB","GNSS"]
- gnss_dev_path="gpsd://localhost:2947"

Host system gpsd process is running but *no GPS device connected*. This mean that gpsd process running OK but not responds any GPS information.

</BR>

## What did you expect? 
I expect the process to be able to handle this "error" situation correctly and not get stuck.
</BR>

## Steps to reproduce this issue

Concentratord starts and begin running various services in loops.

```
2024-11-22T12:35:55.410Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Starting GPS loop
2024-11-22T12:35:55.411Z INFO  [chirpstack_concentratord_sx1302::handler::gps] Starting gpsd reader, server: localhost:2947
2024-11-22T12:35:55.411Z INFO  [libconcentratord::gpsd] Connecting to gpsd, server: localhost:2947
2024-11-22T12:35:55.412Z DEBUG [chirpstack_concentratord_sx1302::handler::stats] Starting stats loop, stats_interval: 30s
2024-11-22T12:35:55.413Z DEBUG [chirpstack_concentratord_sx1302::handler::command] Starting command handler loop
2024-11-22T12:35:55.414Z DEBUG [chirpstack_concentratord_sx1302::handler::uplink] Starting uplink handle loop
2024-11-22T12:35:55.414Z DEBUG [chirpstack_concentratord_sx1302::handler::jit] Starting JIT queue loop
2024-11-22T12:35:55.414Z INFO  [chirpstack_concentratord_sx1302::handler::gps] Starting GPS validation loop

2024-11-22T12:35:55.417Z DEBUG [libconcentratord::gpsd] Version response: {"class":"VERSION","release":"3.25","rev":"3.25","proto_major":3,"proto_minor":15}
2024-11-22T12:35:55.418Z DEBUG [libconcentratord::gpsd] Devices response: {"class":"DEVICES","devices":[]}

```

</BR>

Chipstack server sends  new configuration to concentratord.  Concentratord start stopping service loops...

</BR>

```

2024-11-22T12:36:23.423Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1732278983.423472722s
2024-11-22T12:36:24.423Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1732278984.423755515s
2024-11-22T12:36:25.419Z INFO  [libconcentratord::events] Publishing stats event, rx_received: 1, rx_received_ok: 1, tx_received: 0, tx_emitted: 0
2024-11-22T12:36:25.424Z WARN  [chirpstack_concentratord_sx1302::handler::gps] GPS time reference is not valid, age: 1732278985.42400739s

2024-11-22T12:36:25.430Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] Received stop signal, signal: Configuration
2024-11-22T12:36:25.430Z DEBUG [chirpstack_concentratord_sx1302::handler::gps] GPS validation loop ended

2024-11-22T12:36:25.430Z DEBUG [chirpstack_concentratord_sx1302::handler::jit] Received stop signal, signal: Configuration
2024-11-22T12:36:25.430Z DEBUG [chirpstack_concentratord_sx1302::cmd::root] JIT loop ended
2024-11-22T12:36:25.431Z DEBUG [chirpstack_concentratord_sx1302::handler::stats] Received stop signal, signal: Configuration
2024-11-22T12:36:25.431Z DEBUG [chirpstack_concentratord_sx1302::cmd::root] Stats loop ended
2024-11-22T12:36:25.445Z DEBUG [chirpstack_concentratord_sx1302::handler::uplink] Received stop signal, signal: Configuration
2024-11-22T12:36:25.446Z DEBUG [chirpstack_concentratord_sx1302::cmd::root] Uplink loop ended
2024-11-22T12:36:25.529Z DEBUG [chirpstack_concentratord_sx1302::handler::command] Received stop signal, signal: Configuration
2024-11-22T12:36:25.529Z DEBUG [chirpstack_concentratord_sx1302::cmd::root] Command handle loop ended

```

</BR>

But the ``GPS`` process continues and does not stop.  Although Stop signal! This causes the process to hang.
**debug!("GPS loop ended")** line is missing from the from the log. Why?

</BR> 

In chirpstack-concentratord/chirpstack-concentratord-sx1302/src/cmd/root.rs  run() function hangs.  At the end of the run() function process waits spawned process to be ended. A ``GPS`` process still keep going on and on and on…infinity in next loop: 

```

    for t in threads {
        t.join().unwrap();
    }

```

## A possible reason for the never ending story

TcpStream is opened (in blocking mode!) In chirpstack-concentratord/libconcentratord/src/gpsd.rs module in get_reader() function. Get_reader() returns reader stream and prints following line on the log:

</BR>

```

2024-11-22T12:35:55.417Z DEBUG [libconcentratord::gpsd] Version response: {"class":"VERSION","release":"3.25","rev":"3.25","proto_major":3,"proto_minor":15}
2024-11-22T12:35:55.418Z DEBUG [libconcentratord::gpsd] Devices response: {"class":"DEVICES","devices":[]}
2024-11-22T12:35:55.419Z DEBUG [libconcentratord::gpsd] Watch response: {"class":"WATCH","enable":true,"json":false,"nmea":true,"raw":2,"scaled":false,"timing":false,"split24":false,"pps":false}

```

</BR>

Reader stream is used in chirpstack-concentratord/chirpstack-concentratord-sx1302/src/handler/gps.rs module in  gps_loop() function. Stop signal not handled in gps_loop() ``if let Ok(v) = stop_receive.recv_timeout(Duration::from_millis(0)) { ...``
because gps_reader.read_exact() waits data from gpsd process. Gpsd will not send data to the stream, no GPS module attached to gpsd process. 

Process is stuck in gps_reader.read_exact() call and concentratord process reconfiguration can't be done.

```
        let mut buffer = vec![0; 1];
        gps_reader
            .read_exact(&mut buffer)
            .context("Read from GPS")?;
```

</BR>

This is not a big problem, but annoying because it cause concentratord process hanging.

Async io could be a solution, but I think it is not valid with concentratord architecture?

I dont know does for example  `` TcpStream::stream.set_read_timeout(Duration::millis(5000)) `` help with this issue?

</BR>

## Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v4.9.0  |
|  mqtt-forwarder     |  v4.3.1   |
| Concentratord       |  v4.4.5   |

</BR>
</BR>

But at the end, I am very happy for this gpsd implementation. Someone finally noticed the GPS implementation in the HAL layer ;) 

</BR>

**Thanks, this is a cool platform**

LouneCode, 2024-11-24:
# Never ending story  of the gpsd -  code fix suggestion

Here is a suggestion to make the gpsd process more error tolerant. The repair example is made for the sx1302 module, but of course other modules should also be checked.

After fix the gpsd loop seems to be more stable and error tolerant.

</BR>

## gpsd.rs
[libconcentratord/src/gpsd.rs](https://github.com/chirpstack/chirpstack-concentratord/blob/master/libconcentratord/src/gpsd.rs)

</BR>

Changes:
- add "use std::time::Duration;"
- struct DevicesResponseDevice
  -  Make driver attribute optinal
- add 5 s timeout  to  the reader stream. Reader stream and timeout property is used in [gps.rs](https://github.com/chirpstack/chirpstack-concentratord/blob/master/chirpstack-concentratord-sx1302/src/handler/gps.rs) 
- refactor u-blox driver code 

</BR>

**Code fix**

```
use std::time::Duration;
```
...

```
struct DevicesResponseDevice {
    pub path: String,
    pub driver: Option<String>,
}
```
...

``ublox-check:`` It might be that gpsd process not return driver information. That's why ``DevicesResponseDevice.driver`` should be optional attribute. 

Also debug and serde_json rows are swapped. If a json deserialize error occurs, it will better shows what the gpsd process has respond.

```
    // DEVICES
    let mut b = Vec::new();
    reader.read_until(b'\n', &mut b)?;
    debug!("Devices response: {}", String::from_utf8(b.clone())?);
    let resp: DevicesResponse = serde_json::from_slice(&b)?;

    // WATCH
    let mut b = Vec::new();
    reader.read_until(b'\n', &mut b)?;
    debug!("Watch response: {}", String::from_utf8(b.clone())?);

    for device in &resp.devices {
        if let Some(driver) =  &device.driver {
            if driver == "u-blox" {
                let config_str = format!("&{}=b5620601080001200001010000003294\r\n", device.path);
                debug!("Configuring uBlox device {} for NAV-TIMEGPS", device.path);
                writer.write_all(config_str.as_bytes())?;
                writer.flush()?;
            }
        }
    }
```

</BR>

## gps.rs
[chirpstack-concentratord-sx1302/src/handler/gps.rs](https://github.com/chirpstack/chirpstack-concentratord/blob/master/chirpstack-concentratord-sx1302/src/handler/gps.rs)

</BR>

Changes:
- add "use std::io::ErrorKind;;"
- fine tune code of ther "gps_reader.read_exac" loop. 

 "gps_reader.read_exact" loop must refactored because it can cause the concentratord process to hang during startup when the configuration is loaded loaded but the gpsd process is not ready. (process is blocked by gps_reader.read_exact() function call).

</BR>

**Code fix**
```
use std::io::ErrorKind;
```

```

        match gps_reader.read_exact(&mut buffer).context("Read from GPS") {
            Ok(_) => (),
            Err(err) => {

                // if let Some(e) = err.source() {
                //   debug!("ErroKind: {:?}",e);
                // }

                let mut ek_timeout: Option<ErrorKind> = None;
                for cause in err.chain() {
                    if let Some(io_error) = cause.downcast_ref::<std::io::Error>() {
                        let ek = io_error.kind();
                        if ek == ErrorKind::WouldBlock || ek ==ErrorKind::TimedOut {
                            ek_timeout = Some(ek);
                            break;
                        }
                    }
                }
                if let Some(ek) = ek_timeout {
                    warn!("Attempt to read data from the gpsd timed out.");
                    ()
                } else {
                    return Err(err);
                }
            },
        }
```

</BR>

**After code fixes, Chirpstack-concentratord is like bedrock again**

</BR>

The reason for this issue is that I'll developed tiny miny single core arm A7 platform for concentratord and mqtt-forwarder combination.  Everything was going well until I started use the new ``gpsd`` feature. The system became unstable and I had to investigate where is the problem, in hardware or in software... 

</BR>

The end of the never ending story.

brocaar, 2024-11-25:
Thanks for the extensive report. I will look into this this week. A scenario with `gpsd` without any GNSS device configure was not something I had forseen / tested.

LouneCode, 2024-11-25:
> Lähetetty: maanantai 25. marraskuuta 2024 12.41
> Vastaanottaja: chirpstack/chirpstack-concentratord ***@***.***>
> Kopio: The only Husky in The village ***@***.***>; Author ***@***.***>

Hi,

Thanks and ho hurry,

I have already compiled these suggested changes for my testing purposes. It seems that now system is running stable without complications.  With a multithreaded program, it is always difficult to think of all possible scenarios.

I am currently testing a new small platform for a concentratord + mqtt-forwarder combination and I am using a Luckfox pico ultra W SCB with just one small A7 arm core (32bit) with 256MB memory. The operating system is Alpine Linux 3.20. I am very positively surprised by the very good code performance in this small environment. Thanks for the good quality of CS😉

And good news, I  got the Heltec HT1303 concentratord module to work with the Chirpstack-concentratord without any problems (Not officially suported?)...

BR
--
Tapsa

LouneCode, 2024-11-26:
*"...gpsd without any GNSS device configure was not something I had foreseen / tested..."*

Surprisingly this issue causes a Single Point of Failure in the concentratord and hangs the entire process (in fact, it ends in a kind of deadlock).

The gpsd daemon works fine without a GNSS source and it start working again when GNSS source attached....

commit 31fe20f3 (2024-11-27): Set read timeout + fail if no 'u-blox' device was found.

Closes #198.
closed 2024-11-27 by brocaar